natario1/CameraView #9: Camera preview of resolution 1280x720.

arunmandal53, 2017-08-28:
I have requirement of only displaying and capturing camera on resolution 1280x720. How this possible?

natario1, 2017-08-28:
That's a strange requirement because 

- not all sensor support this resolution
- not all devices are that big

If device is big enough, you can set up a CameraView with fixed width and height (to your values) and use `captureSnapshot()` instead of `capturePicture()`.

arunmandal53, 2017-08-28:
Thank you for reply. Is it possible to set max width for preview. eg. 720p max width. I want to display full screen camera view may be croped but in resolution 1280x720. Help me!

natario1, 2017-08-28:
- It is possible to choose the **preview size** using `CameraView` dimensions.
- It's not possible to choose the **capture size**, sorry. But you can ensure the same aspect ratio of the preview using `cropOutput` (so if preview is 4:3, capture is 4:3 as well).
- If you use `captureSnapshot()`, you will capture a picture with exactly the **preview size**.

I'm not sure if anything fits your case but you can try. I am planning to add more control over the capture size (video already has this) but not pixel-wise as you are requesting, because not all sensors will support that size. Sorry!

closed 2017-08-28 by natario1

PalakSDarji, 2019-07-29:
I want to ask if there is any better option to increase quality in *snapshot()? I want to capture what exactly user is seeing, but I want to have good quality as well.. Please help me!

natario1, 2019-07-29:
@PalakSDarji for resolution you can use setPreviewStreamSize(). But I do not recommend changing this to high values or the camera can fail to open (with Camera2).