natario1/CameraView #12: Image lagging on 4.4.2 lg device

meparam, 2017-08-29:
after image taken it is lagging , not giving smooth scrolling of bottom layout.

natario1, 2017-08-29:
You mean in the demo app? Can you explain better?

meparam, 2017-08-29:
yes when i take the pic  afterword three option coming then in the PicturePreviewActivity when i scroll up it is lagging. please suggest

natario1, 2017-08-29:
I can't reproduce this, but I think it might be due to us calling `startPreview()` from the UI thread after pictures taken. Threading still has to be improved. I will investigate, thanks

meparam, 2017-08-29:
I think size of image is bigger in size that is causing for slow the UI.
Hopping to see update from u soon thanks.

natario1, 2017-08-29:
I have fixed threading in latest version. You can check and let me know

natario1, 2017-09-05:
So, I guess this is just a performance bug in the demo app, that should not load the full image in memory but rather show a smaller preview.

PR #83 (2017-10-28): Improve CameraUtils.decodeBitmap
You can now pass maxWidth and maxHeight to decodeBitmap.

Fixes #76 
Fixes #12
closed 2017-10-28 by natario1