natario1/CameraView #14: Only the original thread that created a view hierarchy can touch its views.

sulisuco, 2017-08-29:
Hello, I'm using your library to capture video, but when I use it, based on the example that comes in it, I get the following error

`Process: com.brive.yooin, PID: 11035
android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
at android.view.ViewRootImpl.checkThread(ViewRootImpl.java:6902)
at android.view.ViewRootImpl.requestLayout(ViewRootImpl.java:1052)
at android.view.View.requestLayout(View.java:19680)
at android.view.View.requestLayout(View.java:19680)
at android.view.View.requestLayout(View.java:19680)
at android.view.View.requestLayout(View.java:19680)
at android.view.View.requestLayout(View.java:19680)
at android.view.View.requestLayout(View.java:19680)
at android.support.constraint.ConstraintLayout.requestLayout(ConstraintLayout.java:1959)
at android.view.View.requestLayout(View.java:19680)
at com.otaliastudios.cameraview.CameraView$CameraCallbacks.onCameraPreviewSizeChanged(CameraView.java:1218)
at com.otaliastudios.cameraview.Camera1.setup(Camera1.java:116)
at com.otaliastudios.cameraview.Camera1.start(Camera1.java:153)
at com.otaliastudios.cameraview.CameraView$2.run(CameraView.java:500)
at android.os.Handler.handleCallback(Handler.java:751)
at android.os.Handler.dispatchMessage(Handler.java:95)
at android.os.Looper.loop(Looper.java:154)
at android.os.HandlerThread.run(HandlerThread.java:61)`

Any recommendation?

natario1, 2017-08-29:
This was fixed in #1 , could you update to latest version? 1.2.0.

natario1, 2017-09-05:
Going to close, feel free to reopen in case you see this again with latest version.

closed 2017-09-05 by natario1

vaidik1011, 2022-04-01:
This problem is still there. if you drawover a compass with animation and then take video/picture snapshot it will show this error..

E/AndroidRuntime: FATAL EXCEPTION: VideoEncoder
    Process: com.vaidik.gpsmapcameraliteforphotolocationtimestamp, PID: 5830
    android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
        at android.view.ViewRootImpl.checkThread(ViewRootImpl.java:6994)
        at android.view.ViewRootImpl.invalidateChildInParent(ViewRootImpl.java:1127)
        at android.view.ViewGroup.invalidateChild(ViewGroup.java:5207)
        at android.view.View.invalidateInternal(View.java:13679)
        at android.view.View.invalidate(View.java:13615)
        at android.view.View.applyLegacyAnimation(View.java:16838)
        at android.view.View.draw(View.java:16918)
        at android.view.ViewGroup.drawChild(ViewGroup.java:3729)
        at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3515)
        at android.view.View.updateDisplayListIfDirty(View.java:16191)
        at android.view.View.draw(View.java:16980)
        at android.view.ViewGroup.drawChild(ViewGroup.java:3729)
        at com.otaliastudios.cameraview.overlay.OverlayLayout.doDrawChild(OverlayLayout.java:180)
        at com.otaliastudios.cameraview.overlay.OverlayLayout.drawChild(OverlayLayout.java:169)
        at android.view.ViewGroup.dispatchDraw(ViewGroup.java:3515)
        at com.otaliastudios.cameraview.overlay.OverlayLayout.drawOn(OverlayLayout.java:151)
        at com.otaliastudios.cameraview.overlay.OverlayDrawer.draw(OverlayDrawer.java:74)
        at com.otaliastudios.cameraview.video.encoding.TextureMediaEncoder.onFrame(TextureMediaEncoder.java:218)
        at com.otaliastudios.cameraview.video.encoding.TextureMediaEncoder.onEvent(TextureMediaEncoder.java:146)
        at com.otaliastudios.cameraview.video.encoding.MediaEncoder$3.run(MediaEncoder.java:244)
        at android.os.Handler.handleCallback(Handler.java:754)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:163)
        at android.os.HandlerThread.run(HandlerThread.java:61)
D/Camera: app passed NULL surface
E/libEGL: call to OpenGL ES API with no current context (logged once per thread)