natario1/CameraView #15: Flipped Images or Mirrored Images

sid-udawat, 2017-08-29:
When taking an Image using the front camera, image appears to be flipped / mirrored. As of now I have handled this in my code using:

```
Matrix m = new Matrix();
m.preScale(-1, 1);
```
But this involves more processing as []bytes are converted to bitmap and again []bytes to perform file storage and displaying purpose.

natario1, 2017-08-29:
There are some TODOs in the code about this, but it's not 100% clear to me what to do. This issue exists in the majority of camera apps (even google stock app) and, for some, could even be desired behavior..

Converting to bitmap just for this is not acceptable. There are special EXIF orientation tags for flipped values, but for that we would need something that can write EXIF tags to a `byte[]` array. Google's support `ExifInterface` won't allow that.

UGhari, 2017-08-30:
This happens because it depends on how an OEM installs the camera module inside the device. They may install the module upside down because of space restrictions we then have to programmatically rotate the view so it looks straight to us. When you take a picture with front facing camera the byte [ ] result doesn't have the EXIF so you can't flip the image directly from the bytes, as you said you will need to make a jpeg and then flip (you can't use EXIF with png). I don't see this being a problem though, wouldn't you want to show the user a preview of the image captured anyway? Then you could just have a save button on that activity? 

Also @natario1 in your feature list you mention camera2 integration, the checkbox isn't checked, but in your source code you have camera2 activity. I haven't looked into it enough within your code but are you not using this code yet for api 21 and above?

natario1, 2017-08-30:
@UGhari are you saying that back cameras return a jpeg `byte[]` while front cameras return a png `byte[]` ? Why would that be?

Yes, for now Camera2 class is not used, just a stub class implementing an interface. We'll get to it!

UGhari, 2017-08-30:
No sorry  I was just stating EXIF data doesn't work with PNG files only JPEG.

sid-udawat, 2017-08-30:
@UGhari @natario1 

Thank you both for the insight, agree that this issue exist on majority of camera apps and many libraries. As of now I have implemented a temporary solution:

- onPictureTakenFront
- onPictureTakenBack
```
private void dispatchOnPictureTaken(byte[] jpeg) {
            final byte[] data = jpeg;
            uiHandler.post(new Runnable() {
                @Override
                public void run() {
                    for (CameraListener listener : mListeners) {

                        switch (mCameraController.getFacing()) {
                            case FRONT:
                                listener.onPictureTakenFront(data);
                                break;
                            case BACK:
                                listener.onPictureTakenBack(data);
                                break;
                        }
                    }
                }
            });
        }
```

```
 @Override
            public void onPictureTakenFront(byte[] jpeg) {
                CameraUtils.decodeByteArray(jpeg, mIsFlipImage, new CameraUtils.ByteArrayCallback() {
                    @Override
                    public void onByteArrayReady(byte[] flipped) {
                       
                    }
                });
}
```
```
public static void decodeByteArray(final byte[] source, final boolean flip, final ByteArrayCallback callback) {
        final Handler ui = new Handler();
        new Thread(new Runnable() {
            @Override
            public void run() {
                final Bitmap bitmap = decodeBitmap(source, flip);
                ui.post(new Runnable() {
                    @Override
                    public void run() {
                        byte[] flipped = getByteArray(bitmap);
                        callback.onByteArrayReady(flipped);
                    }
                });
            }
        }).start();
    }

static byte[] getByteArray(Bitmap bitmap) {
        ByteArrayOutputStream bos = new ByteArrayOutputStream();
        bitmap.compress(Bitmap.CompressFormat.JPEG, 100, bos);
        bitmap.recycle();
        return bos.toByteArray();
    }

    static Pair<Integer, Boolean> getExifOrientation(byte[] source) {
        int orientation;
        boolean flip;
        InputStream stream = null;
        try {
            // http://sylvana.net/jpegcrop/exif_orientation.html
            stream = new ByteArrayInputStream(source);
            ExifInterface exif = new ExifInterface(stream);
            Integer exifOrientation = exif.getAttributeInt(ExifInterface.TAG_ORIENTATION, ExifInterface.ORIENTATION_NORMAL);
            switch (exifOrientation) {
                case ExifInterface.ORIENTATION_NORMAL:
                case ExifInterface.ORIENTATION_FLIP_HORIZONTAL:
                    orientation = 0;
                    break;

                case ExifInterface.ORIENTATION_ROTATE_180:
                case ExifInterface.ORIENTATION_FLIP_VERTICAL:
                    orientation = 180;
                    break;

                case ExifInterface.ORIENTATION_ROTATE_90:
                case ExifInterface.ORIENTATION_TRANSPOSE:
                    orientation = 90;
                    break;

                case ExifInterface.ORIENTATION_ROTATE_270:
                case ExifInterface.ORIENTATION_TRANSVERSE:
                    orientation = 270;
                    break;

                default:
                    orientation = 0;
            }

            flip = exifOrientation == ExifInterface.ORIENTATION_FLIP_HORIZONTAL ||
                    exifOrientation == ExifInterface.ORIENTATION_FLIP_VERTICAL ||
                    exifOrientation == ExifInterface.ORIENTATION_TRANSPOSE ||
                    exifOrientation == ExifInterface.ORIENTATION_TRANSVERSE;

        } catch (IOException e) {
            e.printStackTrace();
            orientation = 0;
            flip = false;
        } finally {
            if (stream != null) {
                try {
                    stream.close();
                } catch (Exception e) {
                }
            }
        }
        return new Pair<>(orientation, flip);
    }

    static Bitmap decodeBitmap(byte[] source) {
        Pair<Integer, Boolean> data = getExifOrientation(source);
        int orientation = data.first;
        boolean flip = data.second;

        Bitmap bitmap = BitmapFactory.decodeByteArray(source, 0, source.length);
        if (orientation != 0 || flip) {
            Matrix matrix = new Matrix();
            matrix.setRotate(orientation);
            Bitmap temp = bitmap;
            bitmap = Bitmap.createBitmap(bitmap, 0, 0, bitmap.getWidth(), bitmap.getHeight(), matrix, true);
            temp.recycle();
        }
        return bitmap;
    }

    static Bitmap decodeBitmap(byte[] source, boolean flip2) {
        Pair<Integer, Boolean> data = getExifOrientation(source);
        int orientation = data.first;
        boolean flip = data.second;

        Bitmap bitmap = BitmapFactory.decodeByteArray(source, 0, source.length);
        if (orientation != 0 || flip) {
            bitmap = processMatrix(bitmap, true, orientation, flip2);
        } else if (flip2) {
            bitmap = processMatrix(bitmap, false, orientation, flip2);
        }
        return bitmap;
    }

    static public Bitmap processMatrix(Bitmap bitmap, boolean isRotate, int orientation, boolean isFlip) {
        Matrix matrix = new Matrix();
        if (isRotate) {
            matrix.setRotate(orientation);
        }
        if (isFlip) {
            matrix.postScale(-1, 1);
        }
        bitmap = Bitmap.createBitmap(bitmap, 0, 0, bitmap.getWidth(), bitmap.getHeight(), matrix, true);
        return bitmap;
    }

  public interface ByteArrayCallback {
        @UiThread
        void onByteArrayReady(byte[] jpeg);
    }
```

This works for me, I know its an "Hack" but seems to be the only working solution as of now.

closed 2019-01-11 by natario1

pedromarthon, 2019-08-21:
Is there any option to flip the result of front camera picture?

PR #360 (2019-01-11): Flip snapshots, add PictureResult.getFacing and VideoResult.getFacing
- Fixes #15 
- Fixes #287 
- Tests: yes