natario1/CameraView #16: Preview Distortion

nonstoplm, 2017-08-30:
preview distortion when camera toggle BACK / FRONT.

UGhari, 2017-08-30:
You will need to give a lot more information than that to debug the problem. Log outputs if possible, model of device, what steps are you taking before this happens.

natario1, 2017-09-04:
@nonstoplm can you provide more information? I can not reproduce this.

Are you testing on a tablet? Which device? Is BACK always good while FRONT is distorted, or what? Which steps are you taking and when do you start seeing the issue?

devalticode, 2017-09-08:
Hi,
I have same issue on Galaxy J5 (SM-J500H).
I set video quality is MAX720p -> Change device orientation + toogle camera facing BACK/FRONT -> issue occurs.
Please help to fix.

natario1, 2017-09-08:
@devalticode are you using latest version?

devalticode, 2017-09-09:
Yes, I did.
Can you please check my layout ? 

[activity_camera.zip](https://github.com/natario1/CameraView/files/1289627/activity_camera.zip)

Johnson145, 2017-09-27:
I got an error fitting to this issue.

The crucial steps to reproduce seem to be:
- `cameraView.setSessionType(SessionType.VIDEO)`
- front camera
- at least one cameraview dimension is using `match_parent` constraint

 Toggling itself does not seem to affect it. It's always the front camera, even if I start it directly in that mode. The back camera is just fine. Furthermore, it's only about the preview. The captured image / video is all good. In my case, distorted means stretched horizontally.

Changing to `cameraView.setSessionType(SessionType.PICTURE)` solves the permanent distortion. Instead there is a temporary distortion whenever the camera is toggled. The image is static and stretched for about a second. Afterwards, everything is fine again. So the picture mode is actually working better.

If I set both dimensions to `wrap_content`, the distortions are gone, too. Of course this is in favor of blank space around the preview.

Tested on a Samsung Galaxy S6 Edge.

Any specific log parts that could help? Right now I don't see anything obvious.

natario1, 2017-09-27:
@Johnson145 thanks. Does this happen with the demo app? Can you start the app directly with the buggy parameters (sessionType = video, dim = match_parent, facing = front) and post logs?

You can post all, we are interested in anything related to `size:` and `onMeasure:`, possibly something else

Johnson145, 2017-09-27:
In order to reproduce the described behavior in your demo app, I need to change the surrounding `ScrollView` to a `LinearLayout`. When using the former, the chosen preview is bigger than the screen and I need to scroll to see everything, but nothing is distorted. Switching to the LinearLayout while keeping the other parameters, compresses the preview in height. The latter seems to be the same distortion as I experience in my app. Furthermore, your demo app gets a black bar simliar to the one in #41 (probably I would see the same bar in my app if I had more space). 

[This is the full log.](https://github.com/natario1/CameraView/files/1338900/error.txt)

natario1, 2017-09-27:
@Johnson145 thanks again, I hope we can sort this out. `camera.isHardwareAccelerated()` returns false, right?

Johnson145, 2017-09-27:
Yes, it does.

natario1, 2017-09-29:
Can you guys check now? Demo app has changed and also a bug was fixed.

Johnson145, 2017-09-30:
Looks good! In the demo app as well as in my own app! Thanks for the changes.

natario1, 2017-09-30:
Ok, closing this, feel free to reopen

closed 2017-09-30 by natario1