natario1/CameraView #20: preview stretch on orientation change

ravikant-dexati, 2017-09-04:
when camera preview is rotated from landscape to portrait get stretchs

natario1, 2017-09-04:
Thanks @ravikant-dexati . So can you help me a bit:

1. Your activity is recreated on orientation changes?
2. You say in landscape it is good, then portrait gets stretched. What happens if you start the activity in portrait mode?
3. Are you testing on a tablet?

ravikant-dexati, 2017-09-04:
I am running the demo app from your git repo .
I am testing on a Motorola's E3 device.
Initially either you start in portrait or landscape it's good but when we
rotate device it stretches.

Best Regards ,
Ravikant Tiwari
Sr. Android Developer
Xymob Inc

natario1, 2017-09-04:
@ravikant-dexati could you add this code and tell me the value of offset?

```java
WindowManager m = (WindowManager) context.getSystemService(Context.WINDOW_SERVICE);
int offset = m.getDefaultDisplay().getRotation()
```

ravikant-dexati, 2017-09-04:
Which class should I add this code

Best Regards ,
Ravikant Tiwari
Sr. Android Developer
Xymob Inc

natario1, 2017-09-04:
Wherever you want, for example in the activity onCreate, using `this` instead of `context`.

ravikant-dexati, 2017-09-04:
The value for rotation is 0 initially

Best Regards ,
Ravikant Tiwari
Sr. Android Developer
Xymob Inc

natario1, 2017-09-04:
If you get 0 I don't know where the issue might be. Will think about it, seems a device specific issue

m1r0n41k, 2017-09-04:
I have same problem on my emulator and on Samsung S6 

in manifest:
` <activity android:name=".app.recorder.MainActivity" />`

in my layout:
`<?xml version="1.0" encoding="UTF-8"?>
<RelativeLayout xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:app="http://schemas.android.com/apk/res-auto"
    android:id="@+id/general"
    android:layout_width="match_parent"
    android:layout_height="match_parent">

    <com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera_container"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:layout_centerInParent="true"
        android:scaleType="centerCrop" />
</RelativeLayout>
`

E/CameraView: onMeasure, surface is not ready. Calling default behavior.
E/CameraView: onMeasure, requested dimensions are (1184[EXACTLY]x720[EXACTLY])
E/CameraView: onMeasure, previewSize is (480.0x640.0)
E/CameraView: onMeasure, both are MATCH_PARENT or fixed value. We adapt. This means CROP_INSIDE. (1184x720)
E/CameraView: onMeasure, requested dimensions are (1184[EXACTLY]x720[EXACTLY])
E/CameraView: onMeasure, previewSize is (480.0x640.0)
E/CameraView: onMeasure, both are MATCH_PARENT or fixed value. We adapt. This means CROP_INSIDE. (1184x720)

m1r0n41k, 2017-09-04:
It works for me only if i use:

`<RelativeLayout xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:tools="http://schemas.android.com/tools"
    android:layout_width="match_parent"
    android:layout_height="match_parent"
    tools:context="com.otaliastudios.cameraview.demo.CameraActivity">

    <ScrollView
        android:layout_width="match_parent"
        android:layout_height="match_parent">

        <com.otaliastudios.cameraview.CameraView
            android:id="@+id/camera"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:scaleType="centerCrop" />
    </ScrollView>

</RelativeLayout>`

natario1, 2017-09-04:
@m1r0n41k thanks. What's the value of `offset` for you when it is distorted? (Read comments above)

m1r0n41k, 2017-09-04:
@natario1  For me mSensorOffset always equals 90 for facing back camera. It's because _cameraInfo.orientation_ - constant value, I think. Now for phones I found hack:

    @Override
    boolean shouldFlipSizes() {
        if (mDeviceOrientation == 90 || mDeviceOrientation == 270) {
            return false;
        } else {
            return mSensorOffset % 180 != 0;
        }
    }

It's only fix my bug with stretch, i didn't check recording and capturing. I think we necessary check orientation for calculate shouldFlipSizes value.

And I found  some problem with _mDeviceOrientation_ - it changed for previous object of Camera1, and no affected for new object. I think something wrong with _OrientationHelper.Callbacks_  in _onDeviceOrientationChanged_, but I need more time for check it.  Now I change _mDeviceOrientation_ for static, it's work, but I we need normal solution. Thanks!

m1r0n41k, 2017-09-04:
@natario1  You can use it for reproduce:

activity_camera.xml
```
<?xml version="1.0" encoding="utf-8"?>
<RelativeLayout xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:tools="http://schemas.android.com/tools"
    android:layout_width="match_parent"
    android:layout_height="match_parent"
    tools:context="com.otaliastudios.cameraview.demo.CameraActivity">

    <com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:scaleType="centerCrop"
        android:layout_width="match_parent"
        android:layout_height="match_parent" />

</RelativeLayout>
```

CameraActivty.java

```
public class CameraActivity extends AppCompatActivity {

    private static String TAG = "CameraActivity_";

    @BindView(R.id.camera)
    CameraView cameraView;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_camera);
        ButterKnife.bind(this);
    }

    @Override
    protected void onResume() {
        super.onResume();
        cameraView.start();
    }

    @Override
    protected void onPause() {
        super.onPause();
        cameraView.stop();
    }

    @Override
    protected void onDestroy() {
        super.onDestroy();
        cameraView.destroy();
    }
}
```

m1r0n41k, 2017-09-04:
And my offser:
```
09-04 23:29:29.467 22579-22579/com.otaliastudios.cameraview.demo D/CameraActivity_: onResume() offset: 0
09-04 23:29:36.168 22579-22579/com.otaliastudios.cameraview.demo D/CameraActivity_: onPause: 
09-04 23:29:36.173 22579-22579/com.otaliastudios.cameraview.demo D/CameraActivity_: onDestroy: 
09-04 23:29:36.357 22579-22579/com.otaliastudios.cameraview.demo D/CameraActivity_: onResume() offset: 1
```

First it for portrait orientation and second for landscape.

natario1, 2017-09-04:
Thanks. That's was I was thinking to, that's why I asked for the device orientation. I think shouldFlipSize() should return (mDeviceOffset + mSensorOffset) % 180 != 0 . Could you check that? :) Or submit a PR with this change if you have time.

m1r0n41k, 2017-09-04:
Yes, I could! 👍

natario1, 2017-09-05:
This should be fixed in the latest `master` branch, will release soon.

closed 2017-09-05 by natario1