natario1/CameraView #21: Dual Camera

freimaks, 2017-09-04:
Hi!

Can I use your library to work with two cameras at the same time?

As a basis I used your example, but I made two CameraView elements: masterCamera and slaveCamera.

Additionally, I changed the code for the button that switches the cameras.

`    
int NextMode = 0;

    @OnClick(R.id.toggleCamera)
    void toggleCamera() {
        if (mCapturingPicture) return;

        masterCamera.stop();
        slaveCamera.stop();

        switch (NextMode) {
            case 0:
                masterCamera.start();
                masterCamera.setFacing(Facing.BACK);
                slaveCamera.setVisibility(View.GONE);
                NextMode = 1;
                break;
            case 1:
                masterCamera.start();
                masterCamera.setFacing(Facing.FRONT);
                slaveCamera.setVisibility(View.GONE);
                NextMode = 2;
                break;
            case 2:
                masterCamera.start();
                masterCamera.setFacing(Facing.BACK);

                slaveCamera.start();
                slaveCamera.setFacing(Facing.FRONT);
                slaveCamera.setVisibility(View.VISIBLE);

                NextMode = 3;
                break;
            case 3:
                masterCamera.start();
                masterCamera.setFacing(Facing.FRONT);

                slaveCamera.start();
                slaveCamera.setFacing(Facing.BACK);
                slaveCamera.setVisibility(View.VISIBLE);
                NextMode = 0;
                break;
        }
    }`

As a result, the application does not always work correctly. After several camera mode switching, an error occurs and the application is closed.

`09-04 12:53:56.284 3787-3837/com.mycom.myapp E/AndroidRuntime: FATAL EXCEPTION: CameraViewController
Process: com.mycom.myapp, PID: 3787
java.lang.RuntimeException: getParameters failed (empty parameters)
at android.hardware.Camera.native_getParameters(Native Method)
at android.hardware.Camera.getParameters(Camera.java:1890)
at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:149)
at com.otaliastudios.cameraview.CameraController$1.run(CameraController.java:48)
at android.os.Handler.handleCallback(Handler.java:739)
at android.os.Handler.dispatchMessage(Handler.java:95)
at android.os.Looper.loop(Looper.java:148)
at android.os.HandlerThread.run(HandlerThread.java:61)`

I run tests on the Nexus 10 emulator (API 23, Android 6).

How to work with two cameras at once? Or is the error due to the emulator?

natario1, 2017-09-04:
I don't know about the emulator and the error is not saying much... Please note that `setFacing` and `start` calls are asynchronous, posted to another thread. Can you try calling things in this order:

- stop()
- setFacing()
- start()

Right not setFacing() is the last. This means that camera opens the old sensor and then tries to switch.

freimaks, 2017-09-04:
I changed it as you said. Yes, apparently this is the correct order of switching. So far there are no errors.

I thought that this is just an emulator error because in reality there is a hardware limitation for working simultaneously with two cameras and I do not really understand whether such a limitation affects the emulator.

natario1, 2017-09-04:
Cool! Let me know if there are other issues. Though if there are known limitations in achieving this, the library is not going to help in any way.

closed 2017-09-04 by natario1

agonshehu, 2017-11-09:
Hello @freimaks  have you tried the dual camera mode in Samsung devices !? i have been having a difficult time with them, as it seems that Samsung intentionally blocks this feature for its own private use.

freimaks, 2017-11-09:
Hello @agonshehu !

No, I tested only on the emulator.
As far as I know the work of two cameras may simply not be supported by the device.

agonshehu, 2017-11-09:
Samsung supports it internally and some claim that they have achieved it. But i unfortunately havent.

asprasan, 2021-01-31:
For dual rear camera phones, when I set ```setFacing(Facing.BACK)``` which of the 2 rear cameras is selected?
Can I select both the rear cameras at once?

benny1611, 2021-05-28:
I'm having the same issue as @asprasan. I'm testing this on a Sony Xperia 10, when I select the back facing camera, I only get one camera, not both