natario1/CameraView #24: Failed to resolve...

antoniovlasic, 2017-09-04:
When I try to bind in the library to my project (Gradle) I get the following error:

```
> Error:Failed to resolve: com.android.support:exifinterface:26.0.1
> <a href="install.m2.repo">Install Repository and sync project</a><br><a href="openFile:C:/Users/Anti/Desktop/PokeEmblem/BabaCamera/app/build.gradle">Open File</a><br><a href="open.dependency.in.project.structure">Show in Project Structure dialog</a>
> 
> Error:Failed to resolve: com.android.support:support-annotations:26.0.1
> <a href="install.m2.repo">Install Repository and sync project</a><br><a href="openFile:C:/Users/Anti/Desktop/PokeEmblem/BabaCamera/app/build.gradle">Open File</a><br><a href="open.dependency.in.project.structure">Show in Project Structure dialog</a>
```

Have absolutly no clue what's the problem :/

natario1, 2017-09-05:
I think you just miss them, can you click on Install Repository and sync project?

antoniovlasic, 2017-09-05:
It doesn't work, I can't even click it

natario1, 2017-09-05:
I think you can search around, it's just about updating your local data.. https://stackoverflow.com/questions/45357000/failed-to-resolve-com-android-supportappcompat-v726-0-0

closed 2017-09-05 by natario1