natario1/CameraView #26: java.lang.RuntimeException: Camera is being used after Camera.release() was called

sw-tt-rahulshah, 2017-09-05:
09-05 12:09:24.102 25020-25020/com.otaliastudios.cameraview.demo E/AndroidRuntime: FATAL EXCEPTION: main
                                                                                   Process: com.otaliastudios.cameraview.demo, PID: 25020
                                                                                   java.lang.RuntimeException: Camera is being used after Camera.release() was called
                                                                                       at android.hardware.Camera.native_setParameters(Native Method)
                                                                                       at android.hardware.Camera.setParameters(Camera.java:1878)
                                                                                       at com.otaliastudios.cameraview.Camera1.startAutoFocus(Camera1.java:693)
                                                                                       at com.otaliastudios.cameraview.CameraView.onGesture(CameraView.java:437)
                                                                                       at com.otaliastudios.cameraview.CameraView.onTouchEvent(CameraView.java:416)
                                                                                       at android.view.View.dispatchTouchEvent(View.java:9300)
                                                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2547)
                                                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2240)
                                                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2553)
                                                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
                                                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2553)
                                                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
                                                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2553)
                                                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
                                                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2553)
                                                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
                                                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2553)
                                                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
                                                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2553)
                                                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
                                                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2553)
                                                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
                                                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2553)
                                                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
                                                                                       at com.android.internal.policy.PhoneWindow$DecorView.superDispatchTouchEvent(PhoneWindow.java:2403)
                                                                                       at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1737)
                                                                                       at android.app.Activity.dispatchTouchEvent(Activity.java:2771)
                                                                                       at android.support.v7.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:68)
                                                                                       at com.android.internal.policy.PhoneWindow$DecorView.dispatchTouchEvent(PhoneWindow.java:2364)
                                                                                       at android.view.View.dispatchPointerEvent(View.java:9520)
                                                                                       at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4230)
                                                                                       at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4096)
                                                                                       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3642)
                                                                                       at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3695)
                                                                                       at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3661)
                                                                                       at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:3787)
                                                                                       at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3669)
                                                                                       at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:3844)
                                                                                       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3642)
                                                                                       at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3695)
                                                                                       at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3661)
                                                                                       at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3669)
                                                                                       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3642)
                                                                                       at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:5922)
                                                                                       at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:5896)
                                                                                       at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:5857)
                                                                                       at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:6025)
                                                                                       at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:185)
                                                                                       at android.os.MessageQueue.nativePollOnce(Native Method)
                                                                                       at android.os.MessageQueue.next(MessageQueue.java:323)
                                                                                       at android.os.Looper.loop(Looper.java:135)
                                                                                       at android.app.ActivityThread.main(ActivityThread.java:5417)

Demo Application crashes every time when we frequently switch application from foreground to background and vice versa.

fuzyfuzi17, 2017-09-12:
Please for god sake someone resolve this issue having same issue

natario1, 2017-09-12:
The stack trace indicates that it happens when you tap on the CameraView in a special occasion. Which special occasion? After `onPause`? While a picture is being taken? While a video is being recorded?

Please add steps to repro.

sw-tt-rahulshah, 2017-09-12:
No I don't even click on capture button. I just move app from foreground to background and vice versa very quickly. Launch app > Click Overview button(3rd button on navigation bar) > Select our demo app again > Click Overview button again.. Continue this process for 2-3 times and it will crash.

natario1, 2017-09-12:
@sw-tt-rahulshah Sorry but I don't get it, the stack trace you posted tells about a touch event inside the CameraView. I can't repro on my device (API 25)

sw-tt-rahulshah, 2017-09-13:
It's happening on Nexus 5, Samsung S4, Moto G4 devices. I don't know about other devices but its happening for sure and other users have same complaints.

natario1, 2017-09-13:
Man it's not that I don't *believe* you, I just need to understand.

@fuzyfuzi17 @sbaar can you post your stacktraces **and** your steps to reproduce the issue?

passos, 2017-09-26:
I got same crash log when I tried to record video in demo application. 

To reproduce the issue, just download and install the demo app, launch it, change the session type, grant the permission, start video recording, then crash.

```
09-26 18:13:25.843 28381 28381 E AndroidRuntime: FATAL EXCEPTION: main
09-26 18:13:25.843 28381 28381 E AndroidRuntime: Process: com.otaliastudios.cameraview.demo, PID: 28381
09-26 18:13:25.843 28381 28381 E AndroidRuntime: java.lang.RuntimeException: Camera is being used after Camera.release() was called
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at android.hardware.Camera.unlock(Native Method)
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at com.otaliastudios.cameraview.Camera1.initMediaRecorder(Camera1.java:648)
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at com.otaliastudios.cameraview.Camera1.startVideo(Camera1.java:604)
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at com.otaliastudios.cameraview.CameraView.startCapturingVideo(CameraView.java:1137)
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at com.otaliastudios.cameraview.CameraView.startCapturingVideo(CameraView.java:1164)
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at com.otaliastudios.cameraview.demo.MainActivity.captureVideo(MainActivity.java:186)
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at com.otaliastudios.cameraview.demo.MainActivity_ViewBinding$4.doClick(MainActivity_ViewBinding.java:82)
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at butterknife.internal.DebouncingOnClickListener.onClick(DebouncingOnClickListener.java:22)
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at android.view.View.performClick(View.java:5204)
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at android.view.View$PerformClick.run(View.java:21153)
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:739)
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:95)
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:148)
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:5417)
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
09-26 18:13:25.843 28381 28381 E AndroidRuntime: 	at de.robv.android.xposed.XposedBridge.main(XposedBridge.java:102)
09-26 18:13:25.844   891  4159 W ActivityManager:   Force finishing activity com.otaliastudios.cameraview.demo/.MainActivity
```

RedMandMs, 2017-09-26:
Yeah... I have same problem. I open activity with CameraView, and then just click button Back quickly, when Camera started work, I get same trace

natario1, 2017-09-26:
@passos if you change session type and you don't have the permission, you get a call to `onCameraClosed`. This means you have to wait for `onCameraOpened` before allowing anything else (like recording video). I know this could be improved.

@RedMandMs can you post your stacktrace as well?

RedMandMs, 2017-09-26:
@natario1 Yeah, I edit my comment above!

RedMandMs, 2017-09-26:
My activity:
```
@BindView(R.id.camera)
    CameraView camera;

    @Override
    protected void onCreate(@Nullable Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_camera);
        ButterKnife.bind(this);
        View decorView = getWindow().getDecorView();
        int uiOptions = View.SYSTEM_UI_FLAG_FULLSCREEN;
        decorView.setSystemUiVisibility(uiOptions);

        camera.addCameraListener(new CameraListener() {

            @Override
            public void onPictureTaken(byte[] jpeg) {
                Toast.makeText(CameraActivity.this, "Я сделяль!", Toast.LENGTH_SHORT).show();
            }
        });
    }

    @Override
    protected void onResume() {
        super.onResume();
        camera.start();
    }

    @Override
    protected void onPause() {
        camera.stop();
        super.onPause();
    }

    @Override
    protected void onDestroy() {
        camera.destroy();
        super.onDestroy();
    }

    @OnClick(R.id.btn_make_photo)
    public void makePhoto() {
        camera.capturePicture();
    }

```

natario1, 2017-09-26:
Will this happen if you invert the two lines inside onPause?

RedMandMs, 2017-09-26:
@natario1 Yeah, sorry, It's just my experiments

RedMandMs, 2017-09-26:
@natario1 And one time now I get this:
```
FATAL EXCEPTION: CameraViewController
                                                                              Process: soft.element.ru.pointandroid, PID: 17529
                                                                              java.lang.RuntimeException: startPreview failed
                                                                                  at android.hardware.Camera.startPreview(Native Method)
                                                                                  at com.otaliastudios.cameraview.Camera1.setup(Camera1.java:136)
                                                                                  at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:163)
                                                                                  at com.otaliastudios.cameraview.CameraController$1.run(CameraController.java:48)
                                                                                  at android.os.Handler.handleCallback(Handler.java:739)
                                                                                  at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                  at android.os.Looper.loop(Looper.java:145)
                                                                                  at android.os.HandlerThread.run(HandlerThread.java:61)
```

RedMandMs, 2017-09-26:
@natario1 
May be activity.xml can help:
```
<RelativeLayout xmlns:android="http://schemas.android.com/apk/res/android"
              xmlns:tools="http://schemas.android.com/tools"
              android:layout_width="match_parent"
              android:layout_height="match_parent"
              android:orientation="vertical"
              tools:context="ru.element.soft.pointandroid.view.activity.CameraActivity">

    <com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:layout_centerInParent="true"
        android:keepScreenOn="true"/>

    <Button
        android:id="@+id/btn_make_photo"
        android:layout_width="130dp"
        android:layout_height="80dp"
        android:background="@drawable/my_button"
        android:layout_marginBottom="10dp"
        android:drawableTop="@android:drawable/ic_menu_camera"
        android:layout_alignParentBottom="true"
        android:layout_centerHorizontal="true"/>

</RelativeLayout>
```

RedMandMs, 2017-09-26:
I think it's relate with RelativeLayout

RedMandMs, 2017-09-26:
I solved my problem, may be it's strange but it's work for me:

```
@BindView(R.id.camera)
    CameraView camera;

    @Override
    protected void onCreate(@Nullable Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_camera);
        ButterKnife.bind(this);
        View decorView = getWindow().getDecorView();
        int uiOptions = View.SYSTEM_UI_FLAG_FULLSCREEN;
        decorView.setSystemUiVisibility(uiOptions);

        camera.addCameraListener(new CameraListener() {
            @Override
            public void onCameraClosed() {
                //Go back only when camera is closed
                CameraActivity.super.onBackPressed();
            }

            @Override
            public void onPictureTaken(byte[] jpeg) {
                Toast.makeText(CameraActivity.this, "Я сделяль!", Toast.LENGTH_SHORT).show();
            }
        });
    }

    @Override
    protected void onResume() {
        super.onResume();
        camera.start();
    }

    @Override
    public void onBackPressed() {
        //On back pressed just stop camera
        camera.stop();
    }

    @Override
    protected void onDestroy() {
        camera.destroy();
        super.onDestroy();
    }

    @OnClick(R.id.btn_make_photo)
    public void makePhoto() {
        camera.capturePicture();
    }
```

natario1, 2017-09-27:
Can anyone test https://github.com/natario1/CameraView/tree/fix26 this branch and see if it still happens?

Johnson145, 2017-10-05:
Got the same error. I implemented your pull request and will keep an eye on it, but it will take a few days to see whether new crash reports are being sent.

Btw, does anyone knows whether the `startPreview failed` error described by @RedMandMs really fits into the `Camera is being used after Camera.release() was called` issue? I get both, but I'm not yet sure about whether it's the same cause.

closed 2017-10-11 by natario1

Johnson145, 2017-10-23:
Your latest changes seem to have solved the very most of the above issues. However, I'm afraid it's still not solved 100%. I got one more error:

`Fatal Exception: com.otaliastudios.cameraview.CameraUnavailableException: Error while starting the camera engine.
       at com.otaliastudios.cameraview.CameraController$2.run(CameraController.java:107)
       at android.os.Handler.handleCallback(Handler.java:743)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:150)
       at android.os.HandlerThread.run(HandlerThread.java:61)
Caused by java.lang.RuntimeException: Camera is being used after Camera.release() was called
       at android.hardware.Camera.native_setParameters(Camera.java)
       at android.hardware.Camera.setParameters(Camera.java:2260)
       at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:224)
       at com.otaliastudios.cameraview.CameraController$2.run(CameraController.java:101)
       at android.os.Handler.handleCallback(Handler.java:743)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:150)
       at android.os.HandlerThread.run(HandlerThread.java:61)`

It's very seldom though.

natario1, 2017-10-24:
@Johnson145 This one you posted might be fixable, can you open a new issue? With more info if you have any

PR #50 (2017-10-11): Don't act on Camera if it is in the middle of a stop
Actions on the internal camera should not take place if the camera is in the middle or is scheduled for a stop / restart.

This hopefully
Fixes #26 
Fixes #31
Might fix also #61 but who knows