natario1/CameraView #28: RuntimeException: Fail to connect to camera service

m1r0n41k, 2017-09-05:
Device: Samsung S6
Android: 5.1.1

Step to reproduce:
1. Open demo app
2. Change screen orientation few times (5-6)

Crash:

```
com.otaliastudios.cameraview.demo D/SensorManager: registerListener :: 0, MPU6500 Acceleration Sensor, 200000, 0,  
com.otaliastudios.cameraview.demo D/SecWifiDisplayUtil: Metadata value : SecSettings2
com.otaliastudios.cameraview.demo W/CameraBase: An error occurred while connecting to camera: 0
system_process V/WindowManager: Adding window Window{1ab62397 u0 d0 com.otaliastudios.cameraview.demo/com.otaliastudios.cameraview.demo.MainActivity} at 2 of 6 (after Window{258291a7 u0 d0 com.google.android.googlequicksearchbox/com.google.android.launcher.GEL})
com.otaliastudios.cameraview.demo D/PhoneWindow: *FMB* isFloatingMenuEnabled mFloatingMenuBtn : null
com.otaliastudios.cameraview.demo D/PhoneWindow: *FMB* isFloatingMenuEnabled return false
system_process I/WindowManager: Screen frozen for +336ms due to AppWindowToken{30159455 token=Token{3b92330c ActivityRecord{3dadce3f u0 com.otaliastudios.cameraview.demo/.MainActivity t491}}}
system_process W/ActivityManager:   Force finishing activity 1 com.otaliastudios.cameraview.demo/.MainActivity
com.otaliastudios.cameraview.demo E/AndroidRuntime: FATAL EXCEPTION: CameraViewController
                                                    Process: com.otaliastudios.cameraview.demo, PID: 14858
                                                    java.lang.RuntimeException: Fail to connect to camera service
                                                        at android.hardware.Camera.<init>(Camera.java:545)
                                                        at android.hardware.Camera.open(Camera.java:385)
                                                        at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:145)
                                                        at com.otaliastudios.cameraview.CameraController$1.run(CameraController.java:48)
                                                        at android.os.Handler.handleCallback(Handler.java:739)
                                                        at android.os.Handler.dispatchMessage(Handler.java:95)
                                                        at android.os.Looper.loop(Looper.java:145)
                                                        at android.os.HandlerThread.run(HandlerThread.java:61)
com.otaliastudios.cameraview.demo E/CameraView: onMeasure, surface is not ready. Calling default behavior.
com.otaliastudios.cameraview.demo E/BufferQueueProducer: [unnamed-14858-1] cancelBuffer: BufferQueue has been abandoned
com.otaliastudios.cameraview.demo E/CameraView: onMeasure, surface is not ready. Calling default behavior.
system_process V/ApplicationPolicy: isApplicationStateBlocked userId 0 pkgname com.otaliastudios.cameraview.demo
com.otaliastudios.cameraview.demo E/BufferQueueProducer: [unnamed-14858-1] cancelBuffer: BufferQueue has been abandoned
com.otaliastudios.cameraview.demo E/BufferQueueProducer: [unnamed-14858-1] cancelBuffer: BufferQueue has been abandoned
com.otaliastudios.cameraview.demo E/BufferQueueProducer: [unnamed-14858-1] cancelBuffer: BufferQueue has been abandoned
com.otaliastudios.cameraview.demo E/BufferQueueProducer: [unnamed-14858-1] cancelBuffer: BufferQueue has been abandoned
com.otaliastudios.cameraview.demo E/BufferQueueProducer: [unnamed-14858-1] cancelBuffer: BufferQueue has been abandoned
com.otaliastudios.cameraview.demo E/BufferQueueProducer: [unnamed-14858-1] cancelBuffer: BufferQueue has been abandoned
```

natario1, 2017-09-05:
On my device the demo app is not going through configuration changes when changing orientation. Can you check if this is fixed using this branch? https://github.com/natario1/CameraView/tree/threads

DEcSENT, 2017-09-05:
I have same issue when using library and change orientation of device.
Device: Meizu M1 Note , Android 4.4.

Demo and threads branch works good.

natario1, 2017-09-05:
@DEcSENT do you mean that the `threads` branch fixes the issue?

DEcSENT, 2017-09-05:
@natario1 Looks like yes. Demo and threads doesn't have this issue.
But application is crashed when i add library to new project and rotate device.

natario1, 2017-09-05:
Sorry, I mean, have you tried applying the `threads` branch to a new project like you say?

DEcSENT, 2017-09-05:
@natario1 Well, when i clone this repository and switch to threads branch in demo all works.
But i want just import library to new project like "compile 'com.otaliastudios:cameraview:1.2.0'" and use camera.
Sorry, if i i don't understand something.

natario1, 2017-09-05:
It's fine, I just wanted to know if there is a problem in `master` (which is like using compile) that was fixed in `threads`. If it is so, I will merge `threads` and release `1.2.1`.

To add the `threads` branch to a new project you should remove the compile line and add a new module pointing to the local cloned repo, after switching to `threads`. I'm sure someone can test this, unfortunately I can't reproduce the bug.

DEcSENT, 2017-09-05:
@natario1 I did this :)
I create new project and create new module (import from cloned repository and switching to threads branch). Now i haven't this issue.

But in landscape orientation i have little trouble - preview is deformate.

natario1, 2017-09-05:
Cool, thanks! So I am going to merge threads.

About deformation, did you have this issue with `compile 1.2.0` or it appeared now?

DEcSENT, 2017-09-05:
@natario1 Yes, have this in both case.

closed 2017-09-05 by natario1

natario1, 2017-09-05:
Everything should be fixed in `1.2.1`.

PR #29 (2017-09-05): WorkerHandler survives config changes
Have handlers survive config changes so start() and stop() calls are streamlined. Fixes #28 .