natario1/CameraView #31: WhiteBalance and HDR error (Camera is being used after Camera.release() was called)

freimaks, 2017-09-05:
Hello!

I found another error. Maybe it's a library error.

An error occurs if we try to switch the HDR or white balance.

I do as you said before:
1. camera.stop();
2. Set the settings (front or back, grid, quality, etc.)
3. camera.start();

Everything works fine if we not use setHDR or setWhiteBalance. It does not matter where we apply the settings (before the camera.start() or after). Using these two settings in onResume does not cause errors.

Here are two pictures with errors:
1. For HDR:
[image]

2. For WhiteBalance:
[image]

natario1, 2017-09-05:
Yeah I can see why this happens. I'll fix this!

natario1, 2017-09-05:
I will consider this low-priority though, because I do not recommended acting this way.

`camera.stop()` and `camera.start()` are not synchronous, and they will be even less so with Camera2.
The correct approach after calling `stop()` is to wait for `onCameraClosed()` to be called, and then start again and/or set parameters.

```java
CameraListener close = new CameraListener() { /* ... listen to onCameraClosed() */ };
camera.addListener(new CameraListener() {
    @Override
    public void onCameraClose() {
        camera.removeListener(this);
        // Set parameters and start again.
        camera.start();
    }
});
camera.stop();
```

Plus, **if all you want is change Facing**, you can call `setFacing()` while the preview is active and the library will manage this for you. Then wait for `onCameraOpened` before changing params.

closed 2017-10-11 by natario1

ManueGE, 2017-10-31:
If I remove listener from `onCameraClose` method, I get a crash:

```
   java.util.ConcurrentModificationException
```

Any other idea?

natario1, 2017-10-31:
It's a bug, can you open a new issue @ManueGE ?

ManueGE, 2017-10-31:
Sure, I will @natario1 . 

Thanks :)

PR #50 (2017-10-11): Don't act on Camera if it is in the middle of a stop
Actions on the internal camera should not take place if the camera is in the middle or is scheduled for a stop / restart.

This hopefully
Fixes #26 
Fixes #31
Might fix also #61 but who knows