natario1/CameraView #41: Black area when cameraview resized squared

Monster-king, 2017-09-12:
[image]

natario1, 2017-09-12:
So you just open app then set these dimensions?

Monster-king, 2017-09-13:
First i created my own app with your library, it has same issue. Then I clone your example, and it has same issue too.

natario1, 2017-09-16:
@Monster-king You didn't answer though :) please use the demo app, then describe clearly the steps to reproduce the bug, then post the logs you see in the logcat. The demo app will log a lot of useful things.

Monster-king, 2017-09-16:
@natario1 At the moment I've installed the app to my Samsung J1 2016 phone, changed the height to 480 (because in width is 480 too). When I pressed the update button I take the message "Height updated! Internal preview size: 640x480". Preview has black area like that screen. Here all logs.
[new 2.txt](https://github.com/natario1/CameraView/files/1308300/new.2.txt)

natario1, 2017-09-16:
@Monster-king thanks!
I just think the camera XML is wrong actually. Can you try modifying it?

If you open `activity_main.xml` you can see that `CameraView` is wrapped in a `RelativeLayout`. That makes no sense. Can you remove the relative layout and try again?

Monster-king, 2017-09-16:
@natario1 Relative layout removed, nothing changed.

natario1, 2017-09-16:
That's disappointing :) I don't know what's wrong right now, but `CameraView` looks to be measuring itself correctly. For the moment I guess you can just avoid to apply runtime changes to width / height. Or if you really need it, ensure you call `requestLayout()` on the view or on its parent.

If you have some time, you could do the following test:

- add `background="@color/red"` to the RelativeLayout
- add `background="@color/green"` to the CameraView
- add to other views if you wish

Then tell me what colour you see in that area that is black now. Just to see what it belongs to. If it's red, then what if you remove the RelativeLayout?

natario1, 2017-09-16:
Nevermind, I was able to reproduce on another device. I will take a look myself

natario1, 2017-09-16:
So, this is a bug with low end devices that have `isHardwareAccelerated()` return false.

In that case `CameraView` will render in a `SurfaceView` (instead of `TextureView`) and we are not able to provide center crop behavior for surface views.

The current workaround for these devices is to use wrap content for both dimensions.

Monster-king, 2017-09-17:
I don't think so, because preview looks like compressed by height. Do you tested it in another device?

natario1, 2017-09-29:
@Monster-king can you try now? Demo app has changed and also a surface bug was fixed.

Monster-king, 2017-10-01:
@natario1 Fixed.

closed 2017-10-01 by natario1