natario1/CameraView #46: distortion bug on orientation change

sanidhya26, 2017-09-17:
I am using the release version 1.2.1....the camera preview is stretched when i start video capture in landscape orientation. Please help

natario1, 2017-09-17:
If this happens with the demo app, can you please

- tell the exact steps you take to reproduce the bug? from app start to when the bug is visible
- post the demo app logs?

sanidhya26, 2017-09-17:
when i start the demo app, I see a black screen instead of preview..so not able to see if the preview get stretched when recording video

sanidhya26, 2017-09-17:
i tried removing the relative layout, the camera view is wrapped in, but still no change..

natario1, 2017-09-17:
Could you post the demo app logs for when you see the black screen?

Also, what happens if you use wrap_content on both dimensions of CameraView?

sanidhya26, 2017-09-17:
09-17 16:12:11.826 305-352/com.otaliastudios.cameraview.demo I/Camera1: onStart: Applying default parameters.
09-17 16:12:11.833 305-305/com.otaliastudios.cameraview.demo W/CameraView: onMeasure: surface is not ready. Calling default behavior.
09-17 16:12:11.838 305-352/com.otaliastudios.cameraview.demo I/Camera1: onStart: Ended
09-17 16:12:11.838 305-352/com.otaliastudios.cameraview.demo I/CameraController: Start: returned from onStart(). Dispatching. STATE_STARTING
09-17 16:12:11.839 305-352/com.otaliastudios.cameraview.demo I/CameraCallbacks: dispatchOnCameraOpened com.otaliastudios.cameraview.CameraOptions@90c979b
09-17 16:12:11.862 305-305/com.otaliastudios.cameraview.demo I/Preview: onSurfaceAvailable: w= 1080 h= 1200
09-17 16:12:11.863 305-305/com.otaliastudios.cameraview.demo I/Camera1: onSurfaceAvailable: Size is 1080x1200
09-17 16:12:11.864 305-352/com.otaliastudios.cameraview.demo I/Camera1: onSurfaceAvailable: Inside handler. About to bind.
09-17 16:12:11.864 305-305/com.otaliastudios.cameraview.demo I/Preview: onSurfaceSizeChanged: w= 1080 h= 1200
09-17 16:12:11.864 305-352/com.otaliastudios.cameraview.demo I/Camera1: setup: Started
09-17 16:12:11.865 305-352/com.otaliastudios.cameraview.demo I/Camera1: shouldFlipSizes: mDeviceOrientation= 0 mSensorOffset= 90
09-17 16:12:11.865 305-352/com.otaliastudios.cameraview.demo I/Camera1: shouldFlipSizes: sensorToDisplay= 90
09-17 16:12:11.867 305-352/com.otaliastudios.cameraview.demo I/Camera1: size: sizesFromList: [3264x2448, 3200x2400, 2592x1944, 2048x1536, 1920x1080, 1600x1200, 1280x960, 1280x768, 1280x720, 1024x768, 800x600, 800x480, 720x480, 640x480, 352x288, 320x240, 176x144]
09-17 16:12:11.867 305-352/com.otaliastudios.cameraview.demo I/Camera1: size: computeCaptureSize: computed 3264x2448
09-17 16:12:11.869 305-352/com.otaliastudios.cameraview.demo I/Camera1: size: sizesFromList: [1920x1080, 1600x1200, 1280x960, 1280x768, 1280x720, 1024x768, 800x600, 800x480, 720x480, 640x480, 352x288, 320x240, 176x144]
09-17 16:12:11.869 305-352/com.otaliastudios.cameraview.demo I/Camera1: size: computePreviewSize: targetRatio: 4:3 surface size: 1080x1200
09-17 16:12:11.870 305-352/com.otaliastudios.cameraview.demo I/Camera1: size: matchSize: found consistent: 6
09-17 16:12:11.870 305-352/com.otaliastudios.cameraview.demo I/Camera1: size: matchSize: found big enough and consistent: 1
09-17 16:12:11.870 305-352/com.otaliastudios.cameraview.demo I/Camera1: size matchSize: returning result 1600x1200
09-17 16:12:11.870 305-352/com.otaliastudios.cameraview.demo I/Camera1: setup: Dispatching onCameraPreviewSizeChanged.
09-17 16:12:11.870 305-352/com.otaliastudios.cameraview.demo I/CameraCallbacks: onCameraPreviewSizeChanged
09-17 16:12:11.870 305-352/com.otaliastudios.cameraview.demo I/Preview: setDesiredSize: desiredW= 1200 desiredH= 1600
09-17 16:12:11.874 305-352/com.otaliastudios.cameraview.demo I/Camera1: setup: Starting preview with startPreview().

sanidhya26, 2017-09-17:
even with the wrap_content on both the dimensions, i see the same black screen

sanidhya26, 2017-09-17:
[image]

natario1, 2017-09-17:
Thanks. You don't see any other log? There should be something about `onMeasure` right after that.

sanidhya26, 2017-09-17:
09-17 16:23:11.241 2138-2138/com.otaliastudios.cameraview.demo I/Preview: crop: applied scaleX= 1.0
09-17 16:23:11.241 2138-2138/com.otaliastudios.cameraview.demo I/Preview: crop: applied scaleY= 1.1999999
09-17 16:23:11.245 2138-2138/com.otaliastudios.cameraview.demo I/Camera1: shouldFlipSizes: mDeviceOrientation= 0 mSensorOffset= 90
09-17 16:23:11.245 2138-2138/com.otaliastudios.cameraview.demo I/Camera1: shouldFlipSizes: sensorToDisplay= 90
09-17 16:23:11.245 2138-2138/com.otaliastudios.cameraview.demo I/CameraView: onMeasure: requested dimensions are (1080[EXACTLY]x1200[EXACTLY])
09-17 16:23:11.245 2138-2138/com.otaliastudios.cameraview.demo I/CameraView: onMeasure: previewSize is (1200.0x1600.0)
09-17 16:23:11.245 2138-2138/com.otaliastudios.cameraview.demo W/CameraView: onMeasure: both are MATCH_PARENT or fixed value. We adapt. This means CROP_CENTER. (1080x1200)
09-17 16:23:11.280 2138-2138/com.otaliastudios.cameraview.demo I/Preview: crop: applied scaleX= 1.0
09-17 16:23:11.280 2138-2138/com.otaliastudios.cameraview.demo I/Preview: crop: applied scaleY= 1.1999999
09-17 16:23:11.280 2138-2138/com.otaliastudios.cameraview.demo I/Camera1: shouldFlipSizes: mDeviceOrientation= 0 mSensorOffset= 90
09-17 16:23:11.280 2138-2138/com.otaliastudios.cameraview.demo I/Camera1: shouldFlipSizes: sensorToDisplay= 90
09-17 16:23:11.282 2138-2138/com.otaliastudios.cameraview.demo I/CameraView: onMeasure: requested dimensions are (1080[EXACTLY]x1200[EXACTLY])
09-17 16:23:11.282 2138-2138/com.otaliastudios.cameraview.demo I/CameraView: onMeasure: previewSize is (1200.0x1600.0)
09-17 16:23:11.282 2138-2138/com.otaliastudios.cameraview.demo W/CameraView: onMeasure: both are MATCH_PARENT or fixed value. We adapt. This means CROP_CENTER. (1080x1200)

natario1, 2017-09-17:
@sanidhya26 sorry man can you post logs from app start to when you don't see any new logs?

sanidhya26, 2017-09-17:
09-17 17:10:13.328 5106-5106/? D/AndroidRuntime: >>>>>> START com.android.internal.os.RuntimeInit uid 2000 <<<<<<
09-17 17:10:13.332 5106-5106/? D/AndroidRuntime: CheckJNI is OFF
09-17 17:10:13.368 5106-5106/? D/ICU: No timezone override file found: /data/misc/zoneinfo/current/icu/icu_tzdata.dat
09-17 17:10:13.378 5118-5118/? D/AndroidRuntime: >>>>>> START com.android.internal.os.RuntimeInit uid 2000 <<<<<<
09-17 17:10:13.382 5118-5118/? D/AndroidRuntime: CheckJNI is OFF
09-17 17:10:13.413 5106-5106/? I/Radio-JNI: register_android_hardware_Radio DONE
09-17 17:10:13.425 5118-5118/? D/ICU: No timezone override file found: /data/misc/zoneinfo/current/icu/icu_tzdata.dat
09-17 17:10:13.433 5106-5106/? D/AndroidRuntime: Calling main entry com.android.commands.wm.Wm
09-17 17:10:13.434 5106-5106/? D/AndroidRuntime: Shutting down VM
09-17 17:10:13.462 5118-5118/? I/Radio-JNI: register_android_hardware_Radio DONE
09-17 17:10:13.481 5118-5118/? D/AndroidRuntime: Calling main entry com.android.commands.am.Am
09-17 17:10:13.482 778-1384/? I/ActivityManager: Force stopping com.otaliastudios.cameraview.demo appid=10096 user=0: from pid 5118
09-17 17:10:13.487 5118-5118/? D/AndroidRuntime: Shutting down VM
09-17 17:10:14.010 778-883/? D/WifiStateMachine: starting scan for "Maverick"WPA_PSK with 2427
09-17 17:10:14.341 5143-5143/? D/AndroidRuntime: >>>>>> START com.android.internal.os.RuntimeInit uid 2000 <<<<<<
09-17 17:10:14.345 5143-5143/? D/AndroidRuntime: CheckJNI is OFF
09-17 17:10:14.386 5143-5143/? D/ICU: No timezone override file found: /data/misc/zoneinfo/current/icu/icu_tzdata.dat
09-17 17:10:14.422 5143-5143/? I/Radio-JNI: register_android_hardware_Radio DONE
09-17 17:10:14.441 5143-5143/? D/AndroidRuntime: Calling main entry com.android.commands.am.Am
09-17 17:10:14.445 778-3052/? I/ActivityManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10000000 cmp=com.otaliastudios.cameraview.demo/.MainActivity} from uid 2000 on display 0
09-17 17:10:14.457 5143-5143/? D/AndroidRuntime: Shutting down VM
09-17 17:10:14.477 778-1304/? I/ActivityManager: Start proc 5161:com.otaliastudios.cameraview.demo/u0a96 for activity com.otaliastudios.cameraview.demo/.MainActivity
09-17 17:10:14.477 5161-5161/? I/art: Late-enabling -Xcheck:jni
09-17 17:10:14.511 5161-5167/? E/art: Failed writing handshake bytes (-1 of 14): Broken pipe
09-17 17:10:14.512 5161-5167/? I/art: Debugger is no longer active
09-17 17:10:14.552 5161-5161/? W/System: ClassLoader referenced unknown path: /data/app/com.otaliastudios.cameraview.demo-1/lib/arm
09-17 17:10:14.576 5161-5161/? W/art: Before Android 4.1, method android.graphics.PorterDuffColorFilter android.support.graphics.drawable.VectorDrawableCompat.updateTintFilter(android.graphics.PorterDuffColorFilter, android.content.res.ColorStateList, android.graphics.PorterDuff$Mode) would have incorrectly overridden the package-private method in android.graphics.drawable.Drawable
09-17 17:10:14.762 5161-5161/? I/CameraCallbacks: onDisplayOffsetChanged 0
09-17 17:10:14.762 5161-5161/? I/CameraController: Start: posting runnable. State: STATE_STOPPED
09-17 17:10:14.762 5161-5181/? I/CameraController: Start: executing. State: STATE_STOPPED
09-17 17:10:14.762 5161-5181/? I/CameraController: Start: about to call onStart() STATE_STARTING
09-17 17:10:14.764 204-16645/? I/CameraService: CameraService::connect call (PID 5161 "com.otaliastudios.cameraview.demo", camera ID 0) for HAL version default and Camera API version 1
09-17 17:10:14.765 204-16645/? I/CameraService: onTorchStatusChangedLocked: Torch status changed for cameraId=0, newStatus=0
09-17 17:10:14.765 204-16645/? W/AudioFlinger: acquireAudioSessionId() unknown client 5161 for session 1487
09-17 17:10:14.771 204-5183/? D/NuPlayerDriver: notifyListener_l(0xb357de40), (1, 0, 0)
09-17 17:10:14.771 204-16645/? W/AudioFlinger: acquireAudioSessionId() unknown client 5161 for session 1488
09-17 17:10:14.772 5161-5184/? D/OpenGLRenderer: Use EGL_SWAP_BEHAVIOR_PRESERVED: true
09-17 17:10:14.785 204-5186/? D/NuPlayerDriver: notifyListener_l(0xb357e7a0), (1, 0, 0)
09-17 17:10:14.785 204-16645/? W/AudioFlinger: acquireAudioSessionId() unknown client 5161 for session 1489
09-17 17:10:14.788 5161-5161/? I/CameraCallbacks: onDisplayOffsetChanged 0
09-17 17:10:14.798 5161-5161/? W/CameraView: onMeasure: surface is not ready. Calling default behavior.
09-17 17:10:14.805 204-5188/? D/NuPlayerDriver: notifyListener_l(0xb357dae0), (1, 0, 0)
09-17 17:10:14.806 204-16645/? I/Camera2ClientBase: Camera 0: Opened. Client: com.otaliastudios.cameraview.demo (PID 5161, UID 10096)
09-17 17:10:14.807 204-16645/? D/mm-camera-intf: mm_camera_open: dev name = /dev/video1, cam_idx = 1
09-17 17:10:14.809 217-217/? I/mm-camera-sensor: module_sensor_start_session:584 session 1
09-17 17:10:14.845 5161-5184/? I/Adreno-EGL: <qeglDrvAPI_eglInitialize:379>: QUALCOMM Build: 10/21/15, 369a2ea, I96aee987eb
09-17 17:10:14.848 5161-5184/? I/OpenGLRenderer: Initialized EGL, version 1.4
09-17 17:10:14.849 217-217/? I/mm-camera: gyro_module_start_session: Enter
09-17 17:10:14.849 217-217/? I/mm-camera: gyro_module_start_session: Init DSPS
09-17 17:10:14.849 217-217/? E/mm-camera: Failed to open sensor1 port
09-17 17:10:14.849 217-217/? I/mm-camera: gyro_module_start_session: dsps_proc_init() failed
09-17 17:10:14.849 217-217/? I/mm-camera: gyro_module_get_port: Exit failure
[... cut ...]
09-17 17:10:31.875 217-5412/? E/mm-camera: port_isp_mct_ctrl_cmd: error in isp_update_buf_info, identity = 0x10002
09-17 17:10:31.875 217-5412/? I/mm-camera: port_isp_mct_ctrl_cmd: X, identity = 0x10002, UPDATE_BUF_INFO, ret = -1
09-17 17:10:31.875 217-5412/? E/mm-camera: port_ispif_proc_mct_ctrl_cmd: forward_event error
09-17 17:10:31.875 217-5412/? E/mm-camera-sensor: port_sensor_port_process_event:1520 failed
09-17 17:10:31.875 217-5412/? E/mm-camera-sensor: module_sensor_module_process_event:2335 failed

sanidhya26, 2017-09-22:
anything?

Johnson145, 2017-09-26:
I'm experiencing distortions when starting in landscape mode, too. Changing the orientation after the camera was loaded in portrait mode works just fine.

@sanidhya26, have you also set your activity's orientation to a fixed value in the manifest (e.g. android:screenOrientation="portrait")? Removing this line seems to fix the problem for me. However, this isn't suitable for me, as I like keeping the activity's orientation while rotating the phone.

Johnson145, 2017-09-26:
66077c1eaab21c21e33866e8c311c2a43ecce86f seems to have solved this issue :)
(at least in my case)

sanidhya26, 2017-09-27:
@Johnson145: yes i have fixed my activity orientation to portrait.

Johnson145, 2017-09-27:
And does the mentioned commit solve your problem, too?

natario1, 2017-10-01:
I think that the latest commit fixed distortion issues. Feel free to reopen.

closed 2017-10-01 by natario1