natario1/CameraView #48: Feature Request: Frame Processor

andrewmunn, 2017-09-20:
I love this library! It's the best camera library I've tested.  Thank you so much for working on this.

The one thing missing is a frame processor so I can do barcode detection and facial recognition. [Fotoapparat](https://github.com/Fotoapparat/Fotoapparat) supports Frame Processing, so I use that, but CameraView supports way more features, like tap-to-focus, so I'd love to switch over.

Thank you

natario1, 2017-09-20:
Thank you for your words.

You mean you need a `byte[]` callback called for each frame?

andrewmunn, 2017-09-20:
Yes, I think that would be perfect

natario1, 2017-09-20:
I'm not against this :) but I don't have much time to work on it. I can help you design this if you want to work on it

natario1, 2017-10-27:
@andrewmunn I'm thinking of providing the byte[] array plus the rotation angle to rotate the image to be consistent with what the user sees. Is there anything else you would like to have in the callback?

I'm also thinking that the callback will be run in a background thread so you can do your job there

PR #82 (2017-10-27): Frame processors
Add APIs and functionality for Frame processors.

- Added `FrameProcessor` interface
- Added `Frame` class
- Added `addFrameProcessor`, `removeFrameProcessor` APIs
- Added tests

Fixes #48 
Fixes #74
closed 2017-10-27 by natario1