natario1/CameraView #49: Specify image resolution

vellrya, 2017-09-24:
Hello! This is one of the best libraries for working with the camera now. But is it possible to specify a resolution for shooting? Now, as far as I understand, the library takes a picture in the maximum possible resolution and cuts it under the resolution of the sides of the device. Is it possible to make the application initially take a picture, for example, with a resolution of 1920 * 1080?

natario1, 2017-09-24:
Current behavior is that you get the picture in the max resolution.

If you fixed some of the CameraView dimensions (for example by setting `match_parent` or a fixed value) **and** cropOutput flag is true, the image will be cropped to match the preview **aspect ratio** (not resolution).

My plan for the future is to let people set something other than `max` in the first place, probably the preferred resolution in megapixels, and a preferred aspect ratio. 

See also #17

vellrya, 2017-09-24:
Thanks for the answer, the fact is that it will take a lot of time to take and process the photo with the maximum resolution and then lower the resolution for my purposes. Therefore, it would be nice to be able to set the resolution in pixels)

natario1, 2017-09-24:
I know, will work on it. For the moment you can take pictures with captureSnapshot. Final size will be the size of the preview.

jesusdanielportillo, 2017-09-29:
I'm using this option in order to get the picture as seen in preview. Gives me an exception:

09-29 12:03:31.736 15476-17046/com.geoubicacion E/AndroidRuntime: FATAL EXCEPTION: CameraViewWorker
                                                                  Process: com.geoubicacion, PID: 15476
                                                                  java.lang.OutOfMemoryError: Failed to allocate a 51121164 byte allocation with 16777216 free bytes and 27MB until OOM
                                                                      at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
                                                                      at android.graphics.Bitmap.nativeCreate(Native Method)
                                                                      at android.graphics.Bitmap.createBitmap(Bitmap.java:939)
                                                                      at android.graphics.Bitmap.createBitmap(Bitmap.java:912)
                                                                      at android.graphics.Bitmap.createBitmap(Bitmap.java:843)
                                                                      at com.otaliastudios.cameraview.CameraUtils.decodeBitmap(CameraUtils.java:137)
                                                                      at com.otaliastudios.cameraview.CropHelper.cropToJpeg(CropHelper.java:24)
                                                                      at com.otaliastudios.cameraview.CameraView$CameraCallbacks$4.run(CameraView.java:1304)
                                                                      at android.os.Handler.handleCallback(Handler.java:739)
                                                                      at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                      at android.os.Looper.loop(Looper.java:145)
                                                                      at android.os.HandlerThread.run(HandlerThread.java:61)

natario1, 2017-09-29:
@jesusdanielportillo CameraUtils.decodeBitmap currently will load the full image in memory. Generally you should load a smaller version. If anyone can open a PR improving `CameraUtils.decodeBitmap`, that would be cool!

https://developer.android.com/topic/performance/graphics/load-bitmap.html

PR #99 (2017-11-14): Add option to choose the picture output size
- Deprecate `toggleFlash`, was broken after #97 
- Deprecate `getCaptureSize()`, replaced with `getPictureSize()`
- Add `cameraView.setPictureSize(SizeSelector)` to choose the output JPEG size
- Add lots of handy size selectors in the `SizeSelectors` class, to apply easy constraints. minWidth, maxWidth, minHeight, maxHeight, minArea, maxArea, smallest, biggest, aspectRatio
- Merge and join selectors with `SizeSelectors.and()` and `SizeSelectors.or()`, very useful
- Add XML attributes for all of this

Closes #49
closed 2017-11-14 by natario1