natario1/CameraView #51: Bug: Capturing videos on Android API < 23 fails

Johnson145, 2017-09-27:
I got another issue: at least when using the Android Emulator, taking videos failes for Lollipop and older. Unfortunately, I can't test it on a real device, but maybe anybode can confirm the bug or restrict it to the Emulator?

Trying to read the created video file, it seems as if it does not contain a single frame. The file itself was created though and has write access.

[image]

[This is the full log.](https://github.com/natario1/CameraView/files/1338925/bug_sdk_22.txt)

Running my own app, I get the same problem, but the [log](https://github.com/natario1/CameraView/files/1338936/bug_sdk_22_own_device.txt) looks a little bit different. The only obviously-related error I can find is `09-27 18:51:50.900 2218-2218/com.siwalusoftware.dogscanner E/MediaRecorder: stop failed: -1007`.

natario1, 2017-09-27:
This should never happen with a real device.

The emulator does a lot of weird stuff when emulating hardware, video recording is officially not supported, there is a small note [here](https://developer.android.com/reference/android/media/MediaRecorder.html) about MediaRecorder.

Out of curiosity, were you able to make this work with some API levels? In title you mention 23 then in the description 21.

Johnson145, 2017-09-27:
Sounds good! 23 and up are working in the emulator just as good as on a real device. 22, 21 and so on do not work.

By the way: THANK YOU VERY MUCH FOR THIS AWESOME LIBRARY! I used to use CameraKit as well and gave up facing all the bugs. This one seems to be way more sophisticated so far. (Didn't plan to spam the issue with this, but just realized for the first time that GitHub does not have a personal message system..)

natario1, 2017-09-29:
Yeah, that looked like a robust base to start but in the end I had to change everything.

Spamming with compliments is allowed :)

closed 2017-09-29 by natario1