natario1/CameraView #58: NullPointerException in startVideo

Johnson145, 2017-10-04:
I'm getting quite a lot NullPointers when users are trying to record a video:

`java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.lock()' on a null object reference
       at com.otaliastudios.cameraview.Camera1.startVideo(Camera1.java:611)
       at com.otaliastudios.cameraview.CameraView.startCapturingVideo(CameraView.java:1153)
       at com.otaliastudios.cameraview.CameraView.startCapturingVideo(CameraView.java:1180)`

So `mCamera` seems to be null here. The `onCameraOpened` event has already been triggered before. Unfortunately, I do not have any extended error logs as I only received the crash reports in Crashlytics. However, it seems to be device and os independent.

natario1, 2017-10-04:
Might be fixed by #50

Johnson145, 2017-10-05:
I'll give it a try. Maybe I'm missing something here, but right now I don't think that #50 will solve the above issue. I'm logging all pause/resume/open/close/etc. events and there is nothing after the camera was opened. So the problem here does not seem to be that someone is trying to record a video while the camera is changing its state.

natario1, 2017-10-05:
That's my feeling, there is a small timeframe while this could happen , and there is a high chance that someone would move to a VIDEO session and start recording too soon.

What else could be causing this? If you have logs to share I'll take a look

Johnson145, 2017-10-05:
Unfortunately, there is really no (visible) user action causing the trouble. Session mode is set to video only once before opening the camera. So this can't be the cause either. I only have my custom logs which do not provide further information. The only thing I can add is that there are usually at least 5 seconds between opening the camera and starting the video record. I'll take a deeper look into the code later on.

natario1, 2017-10-05:
You are right. Actually the only camera.lock() call we do is if we find an exception while starting the video. As if these devices do not support video recording, or `MediaRecorder`, or you passed the wrong file to `startVideo`, or you don't have permissions to write in that location...

Would be cool to reproduce this and know more

Johnson145, 2017-10-05:
It's not permanent. Some of the logged devices successfully recorded a video before. Later on it fails. I guess video recording isn't available temporary, because it already fails on OS level. However, `mCamera` should not be null. I'm now explicitly logging this case to catch further information.

Johnson145, 2017-10-10:
So far I found three different causes for this problem:
1: mCamera == null: false
`Caused by java.lang.IllegalStateException
       at android.media.MediaRecorder.native_start(MediaRecorder.java)
       at android.media.MediaRecorder.start(MediaRecorder.java:870)
       at com.otaliastudios.cameraview.Camera1.startVideo(Camera1.java:608)
       at com.otaliastudios.cameraview.CameraView.startCapturingVideo(CameraView.java:1151)
       at com.otaliastudios.cameraview.CameraView.startCapturingVideo(CameraView.java:1178)`

2: mCamera == null: true
`Caused by java.lang.RuntimeException: start failed.
       at android.media.MediaRecorder.start(MediaRecorder.java)
       at com.otaliastudios.cameraview.Camera1.startVideo(Camera1.java:608)
       at com.otaliastudios.cameraview.CameraView.startCapturingVideo(CameraView.java:1151)
       at com.otaliastudios.cameraview.CameraView.startCapturingVideo(CameraView.java:1178)`

3: mCamera == null: true
`Caused by java.lang.NullPointerException: Attempt to invoke virtual method 'void android.media.MediaRecorder.start()' on a null object reference
       at com.otaliastudios.cameraview.Camera1.startVideo(Camera1.java:608)
       at com.otaliastudios.cameraview.CameraView.startCapturingVideo(CameraView.java:1151)
       at com.otaliastudios.cameraview.CameraView.startCapturingVideo(CameraView.java:1178)`

The [Android manual](https://developer.android.com/reference/android/media/MediaRecorder.html#start()) says something about the first one: `throws IllegalArgumentException if it is called before prepare() or when the camera is already in use by another app.` As `prepare` is called right before `start`, the former case shouldn't be an issue. I guess we can't prevent the latter case and this must be part of permanent error handling. Although I'm wondering how the camera could be used by another app when trying to start a video as the camera should be reserved by CameraView before that point.

I guess the second issue can't be prevent either. However, `mCamera` shouldn't be null there.

The third one is really strange as `start()` triggers the NullPointer, but not the previous call of `prepare()` which uses exactly the same object. Maybe some parallel code unsetting the media recorder in between?

gokrish6670, 2017-10-11:
facing same camera.lock NPE but not frequent and solution is to restart phone... is there any fix

natario1, 2017-10-11:
It's hard to follow 4 traces here, but I am tempted to believe that all of them (caused by calling `startVideo()` in a particular moment, for instance when the camera was stopping / restarting / destroying) were fixed by #50 .

Can you try `1.2.3` and see if you still get these?

Johnson145, 2017-10-11:
I'll give it a shot!

Johnson145, 2017-10-23:
This one seems to be solved, thank you! :)

closed 2017-10-23 by Johnson145