natario1/CameraView #60: Switching SessionType while in another orientation stretches the view

thomasbigger584, 2017-10-09:
Hey,

My requirement is to have 4:3 aspect ratio for picture and fullscreen for video. In portrait this works fine when switching between PICTURE and VIDEO but if I rotate the device to landscape and switch the preview is stretched.

[https://www.dropbox.com/s/z573aa0285lhaov/orientation_change_stretch.mp4?dl=0](Orientation change with setSessionType)

Perhaps is there a way to disable the orientation and keep them in portrait or landscape, or manually set an orientation for his not to be stretched?

Thanks

natario1, 2017-10-09:
Thanks for the report.

How are you achieving 4:3 ratio for picture? Does `CameraView` have fixed dimensions? And then you change them? Or what? What are `layout_width` and `layout_height`?

thomasbigger584, 2017-10-09:
I've extended `CameraView` and override `onMeasure` to set the desired bounds like so: 

    @Override
    protected void onMeasure(int widthMeasureSpec, int heightMeasureSpec) {
        super.onMeasure(widthMeasureSpec, heightMeasureSpec);
        int width = MeasureSpec.getSize(widthMeasureSpec);
        int height = MeasureSpec.getSize(heightMeasureSpec);

        if (0 == mRatioWidth || 0 == mRatioHeight) {
            setMeasuredDimension(width, height);
        } else {

            if (width < height * mRatioWidth / mRatioHeight) {
                setMeasuredDimension(width, width * mRatioHeight / mRatioWidth);
            } else {
                setMeasuredDimension(height * mRatioWidth / mRatioHeight, height);
            }
        }
    }

similar to the google-sample video for the camera2 api: 
https://github.com/googlesamples/android-Camera2Video/blob/master/Application/src/main/java/com/example/android/camera2video/AutoFitTextureView.java

The width is match_parent and it calculates the height accordingly. then on the view switch for video the `layout_height` goes to `match_parent` to make the view take up the whole view

natario1, 2017-10-09:
I see. The internal implementation of onMeasure is a crucial step so if you replace that you're pretty much in the wild.

Right now I would use the layout params to modify this.

```
camera.getLayoutParams().width = screenWidth;
camera.getLayoutParams().height = 4f / 3f * screenWidth;
camera.requestLayout();

// When video...
camera.getLayoutParams().width = MATCH_PARENT;
camera.getLayoutParams().height = MATCH_PARENT;
camera.requestLayout();
```

If you know enough about `onMeasure` you could probably achieve this from inside the view by passing modified width / height specs to `super.onMeasure()`.

thomasbigger584, 2017-10-10:
The issue still persists, it does seem to be something solely when changing orientation and changing the session type, as if the media recorder has the wrong preview dimensions or something when going into landscape.
Would there be any way to turn off the orientation helper in any way so that its always in portrait ?

natario1, 2017-10-10:
The orientation helper in your case only serves the purpose of adding the correct EXIF tag to the result.

We provide no way to change the aspect ratio dynamically, but I'm sure you can sort this out in a couple ways with some trial and error. You could even use two `CameraView`s (just be sure to destroy the first before opening the second - use the right callbacks or use `destroy()` to destroy immediately.

thomasbigger584, 2017-10-10:
Thanks for clearing this up! Ill give it a go

closed 2017-10-23 by natario1