natario1/CameraView #61: startAutoFocus - getParameters or setParameters failed

Johnson145, 2017-10-10:
The `startAutoFocus` causes a crash, because either `getParameters` or `setParameters` failed.

`Fatal Exception: java.lang.RuntimeException: getParameters failed (empty parameters)
       at android.hardware.Camera.native_getParameters(Camera.java)
       at android.hardware.Camera.getParameters(Camera.java:1632)
       at com.otaliastudios.cameraview.Camera1.startAutoFocus(Camera1.java:767)
       at com.otaliastudios.cameraview.CameraView.onGesture(CameraView.java:459)
       at com.otaliastudios.cameraview.CameraView.onTouchEvent(CameraView.java:438)`

and

`Fatal Exception: java.lang.RuntimeException: getParameters failed (empty parameters)
       at android.hardware.Camera.native_getParameters(Camera.java)
       at android.hardware.Camera.getParameters(Camera.java:1632)
       at com.otaliastudios.cameraview.Camera1.startAutoFocus(Camera1.java:767)
       at com.otaliastudios.cameraview.CameraView.onGesture(CameraView.java:459)
       at com.otaliastudios.cameraview.CameraView.onTouchEvent(CameraView.java:438)`

Comes from different devices actually supporting auto focus.

Any thoughts?

natario1, 2017-10-10:
At first I would say it's tapped while the camera is still waiting to be set up (something that #50 should fix when I improve it)

Your stack traces are identical though?

Johnson145, 2017-10-10:
The current version of your changes in #50 is already merged in my case. So this would imply that we really need something like `mScheduledForStop`.

However, the above described errors do not contain anything like `Camera is being used after Camera.release() was called`. Received `{g|s}etParameters failed` from multiple devices and the strack traces are the same.

Johnson145, 2017-10-23:
Unfortunately, these problems could not be solved with your latest changes. The [Android docs state](https://developer.android.com/reference/android/hardware/Camera.html#setParameters(android.hardware.Camera.Parameters)) only that the called methods may fail `if any parameter is invalid or not supported`, but I start to believe that there may be other (hardware-related) issues that can cause these methods to throw a RuntimeException at anytime and that it's not possible to prevent that either.

natario1, 2017-10-24:
@Johnson145 You mean errors triggered by `startAutoFocus` like the stacktrace here? I mean, get/set are called with every API so it's important to know which one was the trigger.

Johnson145, 2017-10-26:
> The Android docs state only that the called methods may fail if any parameter is invalid or not supported, but I start to believe that there may be other (hardware-related) issues that can cause these methods to throw a RuntimeException at anytime and that it's not possible to prevent that either.

The quote above refers to `{g|s}etParameters` in general. I might be wrong though.

In my error logs, the very most of the failed `{g|s}etParameters` calls are related to the auto focus. So I was hoping that there is a specific solution for that cause. However, there may be a more general underlaying cause. I don't know. I'm not changing other parameters during runtime, except toggling the camera face and using the auto focus. So I can't tell for sure whether the `{g|s}etParameters` calls would fail in other situations as often as they do using the auto focus.

What I can tell is that the auto focus is crashing during the normal start (see stack trace above) as well as during the automatically-started reset. Can't find that stack trace right now, but will look for it again.

Johnson145, 2017-10-26:
Sometimes the auto focus fails with a `RuntimeException`, too. However, this is not about `{g|s}etParameters` and may be hardware related. What do you think?

```
Non-fatal Exception: com.otaliastudios.cameraview.CameraConfigurationFailedException: Failed to start auto focus.
       at com.otaliastudios.cameraview.Camera1.startAutoFocus(Camera1.java:933)
       at com.otaliastudios.cameraview.CameraView.onGesture(CameraView.java:459)
       at com.otaliastudios.cameraview.CameraView.onTouchEvent(CameraView.java:438)
       at android.view.View.dispatchTouchEvent(View.java:9399)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2549)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2240)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2555)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2555)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2555)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2555)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2555)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2555)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2555)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2555)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2555)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2555)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2555)
       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2254)
       at com.android.internal.policy.PhoneWindow$DecorView.superDispatchTouchEvent(PhoneWindow.java:2469)
       at com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1785)
       at android.app.Activity.dispatchTouchEvent(Activity.java:2867)
       at android.support.v7.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:68)
       at android.support.v7.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:68)
       at com.android.internal.policy.PhoneWindow$DecorView.dispatchTouchEvent(PhoneWindow.java:2430)
       at android.view.View.dispatchPointerEvent(View.java:9619)
       at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4465)
       at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4331)
       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3865)
       at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3918)
       at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3884)
       at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:4010)
       at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3892)
       at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:4067)
       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3865)
       at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3918)
       at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3884)
       at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3892)
       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3865)
       at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:6164)
       at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:6138)
       at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:6099)
       at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:6280)
       at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:192)
       at android.os.MessageQueue.nativePollOnce(MessageQueue.java)
       at android.os.MessageQueue.next(MessageQueue.java:330)
       at android.os.Looper.loop(Looper.java:137)
       at android.app.ActivityThread.main(ActivityThread.java:5665)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:799)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:689)
Caused by java.lang.RuntimeException: autoFocus failed
       at android.hardware.Camera.native_autoFocus(Camera.java)
       at android.hardware.Camera.autoFocus(Camera.java:1610)
       at com.otaliastudios.cameraview.Camera1.startAutoFocus(Camera1.java:918)
       at com.otaliastudios.cameraview.CameraView.onGesture(CameraView.java:459)
[... cut ...]
       at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:192)
       at android.os.MessageQueue.nativePollOnce(MessageQueue.java)
       at android.os.MessageQueue.next(MessageQueue.java:330)
       at android.os.Looper.loop(Looper.java:137)
       at android.app.ActivityThread.main(ActivityThread.java:5665)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:799)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:689)
```

amakhaev, 2017-11-09:
Hi guys. I have the same problem. Could you tell me please do you want to fix this bug in the (near) future?

P.S. 
The bug is reproduces on device that has android 5.1.1. Another device not catch this exception (android 7.0).
This is my stack trace:
```
java.lang.RuntimeException: setParameters failed
                                                       at android.hardware.Camera.native_setParameters(Native Method)
                                                       at android.hardware.Camera.setParameters(Camera.java:1964)
                                                       at com.otaliastudios.cameraview.Camera1.startAutoFocus(Camera1.java:822)
                                                       at com.otaliastudios.cameraview.CameraView.onGesture(CameraView.java:463)
                                                       at com.otaliastudios.cameraview.CameraView.onTouchEvent(CameraView.java:442)
                                                       at android.view.View.dispatchTouchEvent(View.java:9187)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2680)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2351)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2691)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2366)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2691)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2366)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2691)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2366)
                                                       at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2691)
                                                       at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2366)
                                                       at com.android.internal.policy.impl.PhoneWindow$DecorView.superDispatchTouchEvent(PhoneWindow.java:2743)
                                                       at com.android.internal.policy.impl.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1836)
                                                       at android.app.Activity.dispatchTouchEvent(Activity.java:3043)
                                                       at com.android.internal.policy.impl.PhoneWindow$DecorView.dispatchTouchEvent(PhoneWindow.java:2704)
                                                       at android.view.View.dispatchPointerEvent(View.java:9397)
                                                       at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:5082)
                                                       at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4899)
                                                       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4356)
                                                       at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4409)
                                                       at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4375)
                                                       at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:4501)
                                                       at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4383)
                                                       at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:4558)
                                                       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4356)
                                                       at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4409)
                                                       at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4375)
                                                       at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4383)
                                                       at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4356)
                                                       at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:6979)
                                                       at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:6852)
                                                       at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:6823)
                                                       at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:7069)
                                                       at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:185)
                                                       at android.os.MessageQueue.nativePollOnce(Native Method)
[... cut ...]
                                                     at android.app.ActivityThread.main(ActivityThread.java:7007)
                                                     at java.lang.reflect.Method.invoke(Native Method)
                                                     at java.lang.reflect.Method.invoke(Method.java:372)
                                                     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1404)
                                                     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1199)
```

Johnson145, 2017-11-10:
Btw, I found the stacktraces for the automatically-scheduled focus reset again. That code part is failing, too. It's actually even the place that fails most often.

natario1, 2017-11-10:
@Johnson145 thanks! Can you post the stacktrace?

Johnson145, 2017-11-10:
This one is almost identical to the ones above. It's raised by Samsung devices only:
```
Non-fatal Exception: com.otaliastudios.cameraview.CameraConfigurationFailedException: Failed to reset auto focus.
       at com.otaliastudios.cameraview.Camera1$1.run(Camera1.java:58)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:158)
       at android.os.HandlerThread.run(HandlerThread.java:61)
Caused by java.lang.RuntimeException: setParameters failed
       at android.hardware.Camera.native_setParameters(Camera.java)
       at android.hardware.Camera.setParameters(Camera.java:1984)
       at com.otaliastudios.cameraview.Camera1$1.run(Camera1.java:51)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:158)
       at android.os.HandlerThread.run(HandlerThread.java:61)
```

This one differs a bit and is raised by arbitrary devices (including some Samsung models, too):
```
Non-fatal Exception: com.otaliastudios.cameraview.CameraConfigurationFailedException: Failed to reset auto focus.
       at com.otaliastudios.cameraview.Camera1$1.run(Camera1.java:58)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:148)
       at android.os.HandlerThread.run(HandlerThread.java:61)
Caused by java.lang.RuntimeException: cancelAutoFocus failed
       at android.hardware.Camera.native_cancelAutoFocus(Camera.java)
       at android.hardware.Camera.cancelAutoFocus(Camera.java:1331)
       at com.otaliastudios.cameraview.Camera1$1.run(Camera1.java:45)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:148)
       at android.os.HandlerThread.run(HandlerThread.java:61)
```

Johnson145, 2017-11-10:
The very last stacktrace seems to occur only while recording a video.

PR #97 (2017-11-14): Make CameraView thread-safe
- Make CameraView completely thread-safe
- All actions are posted to a single thread, like Camera1 docs suggest
- This should avoid any "used after release" error
- Add `CameraListener.onCameraError()` and `CameraException` for unrecoverable errors
- Fix some bugs with video recording and MediaRecorder not correctly setup or released

Fixes #77 
Fixes #61
closed 2017-11-14 by natario1