natario1/CameraView #63: Camera view does not fill the screen

Isquierdo, 2017-10-11:
I'm having some problems with the CameraView size, in some devices it does not fill all the screen, showing the background sides or bottom/top depending on the device.

```
<RelativeLayout xmlns:android="http://schemas.android.com/apk/res/android"
        xmlns:app="http://schemas.android.com/apk/res-auto"
        android:layout_width="match_parent"
        android:layout_height="match_parent">

        <com.otaliastudios.cameraview.CameraView
            android:id="@+id/camera_view"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:layout_alignParentBottom="true"
            android:layout_alignParentEnd="true"
            android:layout_alignParentLeft="true"
            android:layout_alignParentRight="true"
            android:layout_alignParentStart="true"
            android:layout_alignParentTop="true"
            android:keepScreenOn="true"
            app:cameraSessionType="video" />
    </RelativeLayout>
```

natario1, 2017-10-11:
Would this happen without RelativeLayout? Are you doing anything strange?

Isquierdo, 2017-10-13:
@natario1 I found the problem in my project Manifest, the hardwareAccelerated was "false".
Just changed to "true" and now it's working fine.

closed 2017-10-13 by Isquierdo

softmarshmallow, 2019-01-27:
why would this happen?