natario1/CameraView #64: Stretching out image

Talgat777, 2017-10-12:
Hey,

We are having an issue with Camera view stretching and not showing in full screen for Galaxy tab 4 android version 5.0.2.

On other devices tested so far the view works fine.

See the screenshot

[image]

natario1, 2017-10-12:
Can you reproduce this inside the demo app? If so, can you post logs from the buggy device?

Talgat777, 2017-10-12:
It doesn't stretch when running demo. What can we do wrong?
  
[image]

natario1, 2017-10-12:
I don't know. Try to put it out of useless containers, avoid messy stuff (overriding `onMeasure` or similar)

Talgat777, 2017-10-12:
<?xml version="1.0" encoding="utf-8"?>
<RelativeLayout xmlns:android="http://schemas.android.com/apk/res/android"
    android:layout_width="match_parent"
    android:layout_height="match_parent"
    xmlns:tools="http://schemas.android.com/tools"
    android:background="@color/overlay"
    xmlns:app="http://schemas.android.com/apk/res-auto"
    android:orientation="vertical">

    <com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:keepScreenOn="true"
        app:cameraFacing="back"
        app:cameraJpegQuality="100"
        app:cameraWhiteBalance="auto"
        app:cameraHdr="off"
        app:cameraSessionType="picture"
        android:layout_width="match_parent"
        android:layout_height="match_parent" />

    <ImageView
        android:id="@+id/result"
        android:layout_width="100dp"
        android:layout_height="100dp"
        android:layout_gravity="bottom"
        android:layout_margin="10dp"
        android:scaleType="centerCrop" />

    <ImageButton
        android:id="@+id/ib_library"
        android:layout_width="56dp"
        android:layout_height="56dp"
        android:layout_alignParentBottom="true"
        android:layout_alignParentRight="true"
        android:layout_margin="10dp"
        android:background="@drawable/background"
        android:src="@drawable/library_button" />

    <ImageButton
        android:id="@+id/ib_take_photo"
        android:layout_width="56dp"
        android:layout_height="56dp"
        android:layout_alignParentBottom="true"
        android:layout_centerHorizontal="true"
        android:layout_margin="10dp"
        android:background="@drawable/background"
        app:srcCompat="@drawable/ic_photo" />

    <TextView
        android:id="@+id/no_permission"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content"
        android:layout_gravity="center_horizontal|bottom"
        android:layout_marginBottom="50dp"
        android:layout_alignParentBottom="true"
        android:layout_centerHorizontal="true"
        android:text="No Camera permission granted."
        android:textSize="20sp"
        android:visibility="gone"
        tools:visibility="visible" />

    <ImageButton
        android:id="@+id/switchCamera"
        android:layout_width="56dp"
        android:layout_height="56dp"
        android:background="@drawable/background"
        app:srcCompat="@drawable/ic_switch"
        android:layout_gravity="top|right"
        android:layout_alignParentTop="true"
        android:layout_alignParentRight="true"
        android:layout_margin="10dp"/>

</RelativeLayout>

natario1, 2017-10-12:
It looks like it won't work well inside RelativeLayouts, see #63 also

Talgat777, 2017-10-12:
Front camera image is upside down though

[image]

natario1, 2017-10-13:
@Talgat777 can you open a new issue with that? And close this one if it is solved.

In the new issue please add steps to reproduce the issue in the demo app, and demo app logs until the moment the issue is visible

natario1, 2017-11-19:
It's been a month now, I'm closing this. Feel free to open a new issue if the problem persist with latest version, and filling the issue form.

closed 2017-11-19 by natario1

saidurcse, 2018-10-01:
I have reproduced the issue in demo app. Please let me know do you need any more information regarding this issue?