natario1/CameraView #65: Video timer inconsistencies

UGhari, 2017-10-13:
I know you have already closed an issue similar to this, but I think you would understand best how to approach such an issue as you know how all of your code works. As the title says after hitting record I show a time elapsed. I've tried to use a Handler:

//for 10 seconds 
`new CountDownTimer(10000, 1000) {

     public void onTick(long millisUntilFinished) {
         timerVideo.setText("seconds remaining: " + millisUntilFinished / 1000);
     }

     public void onFinish() {
   
     }
  }.start();`

I have also used CountDownTimer:

`new CountDownTimer(10000, 1000) {

                        public void onTick(long millisUntilFinished) {
                            forcePressNext = true;
                            timerVideo.setText(20-millisUntilFinished/1000 + "s");
                        }

                        public void onFinish() {
                            timerVideo.setText("10s!");
                            if (forcePressNext == true) {
                                isRecording = false;
                                cameraView.stopCapturingVideo();
                            }
                        }
                    }.start();`

Both are quite inconsistent by 2/3s and the inconsistency gets larger on older devices, have you got any ideas?

thanks

natario1, 2017-10-13:
No, surely there is a delay between the moment you call `startVideo` and the moment the recording actually start (we must setup recording), and between `endVideo` and the moment it actually ends.

I don't know if and how this could be improved.

UGhari, 2017-10-13:
I shall investigate further and let you know if I find anything

Barai-Vivek, 2018-01-10:
This is not a good way but i solved my problem.
It did work for me in when i tried to 1500 milliseconds to my desired seconds 
for example if you want 5 sec video you need (5000+1500) milliseconds it worked for me in motorola and asus devices.

                long capturesec =  (5*1000)+1500;
                //for 5 sec = 6500
                //for 10 sec 11500
                //for 15 sec 16500
               cameraView.startCapturingVideo(file);
                new CountDownTimer(capturesec, 1000) {
                    public void onTick(long millisUntilFinished) {
                        txtTime.setText("00:0" + ((millisUntilFinished-1500) / 1000));
                    }

                    public void onFinish() {
                        txtTime.setText("00:00");
                        cameraView.stopCapturingVideo();
                      
                    }
                }.start();

Aks-4125, 2018-01-22:
@natario1  any update for this issue? I am still facing same issue. video duration is different than countdown timer.

PR #172 (2018-03-01): Add setVideoMaxDuration() API
Fixes #65 (it's the best we can do)
closed 2018-03-01 by natario1