natario1/CameraView #70: integration issue

tabassumLatif, 2017-10-19:
Error:Execution failed for task ':app:processDebugManifest'.
> Manifest merger failed with multiple errors, see logs

natario1, 2017-10-23:
So see logs?

RobertoMorelos, 2017-10-24:
I'm facing the same error:
Error:Execution failed for task ':app:processDebugManifest'.
> Manifest merger failed : Attribute meta-data#android.support.VERSION@value value=(26.1.0) from [com.android.support:cardview-v7:26.1.0] AndroidManifest.xml:25:13-35
  	is also present at [com.android.support:exifinterface:26.0.2] AndroidManifest.xml:25:13-35 value=(26.0.2).
  	Suggestion: add 'tools:replace="android:value"' to <meta-data> element at AndroidManifest.xml:23:9-25:38 to override.

RobertoMorelos, 2017-10-24:
Apparently there are different versions of the android support library, I'm using the latest one (26.1.0) but the library is using the 25.

v-gar, 2017-10-25:
Hello, same problem here. @natario1 I already wrote you an email about this topic on Monday.

natario1, 2017-10-25:
I think you can compile using `exclude group: 'com.android.support'` to fix this. Anyway if someone wants to PR and upgrade this to 26.1.0, that's welcome!

v-gar, 2017-10-25:
Opened #73

natario1, 2017-10-25:
Merged, thanks

closed 2017-10-25 by natario1

v-gar, 2017-10-25:
Can you push it to the Gradle repo?

RobertoMorelos, 2017-10-25:
If we exclude 'com.android.support' some features don't work properly like the cameraCropOutput and CameraUtils (I got some crashes). @natario1 in which version this upgrade/fix will be at? Thanks a lot!

natario1, 2017-10-25:
Oh sure. I will release a new version in the next days, meanwhile you can use some other workaround, just google the error to find. (or you can use 26.0.1 in your other dependencies since there's no real difference). Thanks!

v-gar, 2017-10-29:
Works now. (v1.3.0) :+1: