natario1/CameraView #74: Stock snapshot/image byte array

VyacheslavShmakin, 2017-10-25:
Could you please add some functionality to provide return stock byte array from snapshot/image?
I think it would be much better if user will have ability decide what to do with this byte array.
It may be some kind of option. 
Like captureImage(boolean processImage) and captureSnapshot(boolean processImage)

natario1, 2017-10-25:
What's the use case for this? You can set `cropOutput` to false and that's it.

I think you might be satisfied with a frame processor #48

VyacheslavShmakin, 2017-10-25:
Oh. It seems I missed #48. But there is some details.
There is some modifications for byte array that recieved from taken frame in **Camera1** class.
`
final boolean consistentWithView = (sensorToDevice + sensorToDisplay + 180) % 180 == 0;
                        byte[] rotatedData = RotationHelper.rotate(data, preWidth, preHeight, sensorToDevice);
                        YuvImage yuv = new YuvImage(rotatedData, format, postWidth, postHeight, null);
                        mCameraCallbacks.processSnapshot(yuv, consistentWithView, exifFlip);`

So, I think it would be nice to have original byte array as additional parameter.
By the way, thank you for this great library:)

PR #82 (2017-10-27): Frame processors
Add APIs and functionality for Frame processors.

- Added `FrameProcessor` interface
- Added `Frame` class
- Added `addFrameProcessor`, `removeFrameProcessor` APIs
- Added tests

Fixes #48 
Fixes #74
closed 2017-10-27 by natario1