natario1/CameraView #76: decodeBitmap utility OOM with large pictures

ninjaintrouble, 2017-10-26:
`FATAL EXCEPTION: FallbackCameraThread
                                                                          Process:  camera_module, PID: 27112
                                                                          java.lang.OutOfMemoryError: Failed to allocate a 63701004 byte allocation with 16777216 free bytes and 60MB until OOM
                                                                              at dalvik.system.VMRuntime.newNonMovableArray(Native Method)
                                                                              at android.graphics.Bitmap.nativeCreate(Native Method)
                                                                              at android.graphics.Bitmap.createBitmap(Bitmap.java:975)
                                                                              at android.graphics.Bitmap.createBitmap(Bitmap.java:946)
                                                                              at android.graphics.Bitmap.createBitmap(Bitmap.java:877)
                                                                              at com.otaliastudios.cameraview.CameraUtils.decodeBitmap(CameraUtils.java:137)
                                                                              at com.otaliastudios.cameraview.CameraUtils$1.run(CameraUtils.java:74)
                                                                              at android.os.Handler.handleCallback(Handler.java:739)
                                                                              at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                              at android.os.Looper.loop(Looper.java:158)
                                                                              at android.os.HandlerThread.run(HandlerThread.java:61)`

Maybe a lossless rotation could be done in native or with renderscript ?

ninjaintrouble, 2017-10-26:
It crashes in the demo app too.

Johnson145, 2017-10-26:
`decodeBitmap` loads the full bitmap into memory which will force an OOM at some point. If you don't need the full resolution, you can load a smaller bitmap like so:

```
/**
 * Load an image bitmap from a given uri.
 * Image might be down-sampled, if it is really big.
 * When changing this method, have a look at loadBitmap(String filePath), too.
 * @param contentUri
 * @param minWidth If possible, the image will be resized such that minWidth and minHeight are kept.
 * @param minHeight If possible, the image will be resized such that minWidth and minHeight are kept.
 * @return
 */
public static Bitmap loadBitmap(@NonNull final Uri contentUri, int minWidth, int minHeight) {
    // validate parameter
    Validation.ensureNonNull(contentUri, "Can not load a bitmap from a null uri.");

    // load bitmap from uri and pay attention to possible down-sampling
    Bitmap resultBitmap;
    InputStream inputStream = null;
    try {
        inputStream = MainApp.getAppContext().getContentResolver().openInputStream(contentUri);

        // First decode with inJustDecodeBounds=true to add dimensions to the options object
        final BitmapFactory.Options options = new BitmapFactory.Options();
        options.inJustDecodeBounds = true;
        BitmapFactory.decodeStream(inputStream, null, options);

        // the stream position is changed now, we need to close and re-open it
        inputStream.close();

        // Calculate inSampleSize
        options.inSampleSize = calculateBitmapInSampleSize(options,
                minWidth, minHeight);
        if (options.inSampleSize != 1) {
            Crashlytics.log(Log.INFO, TAG, "Down-sampling image by a factor of " + options.inSampleSize + " from " + options.outWidth + " x " + options.outHeight);
        }
        else {
            Crashlytics.log(Log.INFO, TAG, "No Down-sampling. Loading the full image (" + options.outWidth + " x " + options.outHeight + ")");
        }

        // re-open stream at the beginning
        inputStream = MainApp.getAppContext().getContentResolver().openInputStream(contentUri);

        // Decode bitmap with inSampleSize set to reduce required memory
        // (in case of an error resultBitmap will be null, but no exception will be thrown natively)
        options.inJustDecodeBounds = false;
        resultBitmap = BitmapFactory.decodeStream(inputStream, null, options);
        if (resultBitmap == null) {
            throw new DecodeBitmapStreamFailed(inputStream);
        }

    } catch (FileNotFoundException e) {
        resultBitmap = null;
        Crashlytics.log(Log.ERROR, TAG, "The image file (uri) " + contentUri + " does not exist.");
        Crashlytics.logException(e);
    } catch (IOException e) {
        resultBitmap = null;
        Crashlytics.log(Log.ERROR, TAG, "Could not read bitmap from uri, because the input stream could not be closed: " + e);
        Crashlytics.logException(e);
    } catch (DecodeBitmapStreamFailed e) {
        resultBitmap = null;
        Crashlytics.log(Log.ERROR, TAG, "Could not read bitmap from uri, because " + e.getMessage());
        Crashlytics.logException(e);
    }

    // close input stream, if it is still open
    if( inputStream != null ) {
        try {
            inputStream.close();
        } catch (IOException e) {
            // this error doesn't really matter, but we should log it at least.
            Crashlytics.log(Log.WARN, TAG, "Could not close input stream: " + e);
        }
    }

    checkBitmapResult(resultBitmap, contentUri.toString());

    return resultBitmap;
}

/**
 * Check whether bitmap is a valid Bitmap object and log the results.
 * @param bitmap
 */
private static void checkBitmapResult(Bitmap bitmap, String source) {
    if (bitmap == null) {
        Crashlytics.log(Log.ERROR, TAG, "Could not load bitmap from " + source);
    }
    else {
        int bytes;
        if (Build.VERSION.SDK_INT > 18) {
            bytes = bitmap.getAllocationByteCount();
        }
        else {
            bytes = bitmap.getByteCount();
        }
        Crashlytics.log(Log.INFO, TAG, "Successfully loaded bitmap (" + humanReadableByteCount(bytes, true) + ", " + bitmap.getWidth() + " x " + bitmap.getHeight() + "px) from " + source);
    }
}
```

You may need to modify it in order to provide your correct context as well as removing unneccessary log calls.

natario1, 2017-10-26:
See also 
#12
https://github.com/natario1/CameraView/issues/49#issuecomment-333179061

PR #83 (2017-10-28): Improve CameraUtils.decodeBitmap
You can now pass maxWidth and maxHeight to decodeBitmap.

Fixes #76 
Fixes #12
closed 2017-10-28 by natario1