natario1/CameraView #78: Error while stopping the camera engine - NPE

Johnson145, 2017-10-26:
Sometimes I get a `NullPointerException` when trying to stop the camera:

`Fatal Exception: com.otaliastudios.cameraview.CameraUnavailableException: Error while stopping the camera engine.
       at com.otaliastudios.cameraview.CameraController$3.run(CameraController.java:134)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:168)
       at android.os.HandlerThread.run(HandlerThread.java:61)
Caused by java.lang.NullPointerException: Attempt to write to field 'boolean android.hardware.Camera.mIsReleased' on a null object reference
       at android.hardware.Camera.releaseStaticSubCam(Camera.java:833)
       at android.hardware.Camera.release(Camera.java:804)
       at com.otaliastudios.cameraview.Camera1.onStop(Camera1.java:255)
       at com.otaliastudios.cameraview.CameraController$3.run(CameraController.java:128)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:168)
       at android.os.HandlerThread.run(HandlerThread.java:61)`

I don't think this is the same problem as #77, because here the camera was started completely. At least I received the `onCameraOpened` event before getting this error.

natario1, 2017-10-26:
This is expected, it should not be an issue, it's in a try block.

See, when app is closed we release the camera twice, one from onPause, one from onDestroy, just to avoid some nasty bugs when changing configurations. This means that one of the two will fail (but we are sure that camera is released). The exception is caught anyway

natario1, 2017-10-27:
So from my understanding this should be a non-issue related to destroy, feel free to reopen if it's not the case.

closed 2017-10-27 by natario1