natario1/CameraView #79: <merge /> can be used only with a valid ViewGroup root and attachToRoot=true

Aissa-H, 2017-10-26:
Hi @natario1 !

After passing to the version 3.0 of Android Studio and to the version 27 of buildTools and support library, I had this crash immediately when I open the activity the CameraView is into. Maybe you have an idea about that? 

Thank you for your work! :+1: 

```
E/AndroidRuntime: FATAL EXCEPTION: main
                  Process: fr.appsolute.teazit.dev, PID: 4772
                  android.view.InflateException: <merge /> can be used only with a valid ViewGroup root and attachToRoot=true
                      at android.view.LayoutInflater.inflate(LayoutInflater.java:475)
                      at android.view.LayoutInflater.inflate(LayoutInflater.java:414)
                      at com.otaliastudios.cameraview.TextureCameraPreview.onCreateView(TextureCameraPreview.java:24)
                      at com.otaliastudios.cameraview.TextureCameraPreview.onCreateView(TextureCameraPreview.java:13)
                      at com.otaliastudios.cameraview.CameraPreview.<init>(CameraPreview.java:37)
                      at com.otaliastudios.cameraview.TextureCameraPreview.<init>(TextureCameraPreview.java:18)
                      at com.otaliastudios.cameraview.CameraView.instantiatePreview(CameraView.java:168)
                      at com.otaliastudios.cameraview.CameraView.instantiatePreview(CameraView.java:174)
                      at com.otaliastudios.cameraview.CameraView.onAttachedToWindow(CameraView.java:184)
                      at android.view.View.dispatchAttachedToWindow(View.java:13926)
                      at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:2882)
                      at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:2890)
                      at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:2890)
                      at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:2890)
                      at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:2890)
                      at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:2890)
                      at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:2890)
                      at android.view.ViewGroup.dispatchAttachedToWindow(ViewGroup.java:2890)
                      at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:1311)
                      at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1073)
                      at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:5988)
                      at android.view.Choreographer$CallbackRecord.run(Choreographer.java:767)
                      at android.view.Choreographer.doCallbacks(Choreographer.java:580)
                      at android.view.Choreographer.doFrame(Choreographer.java:550)
                      at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:753)
                      at android.os.Handler.handleCallback(Handler.java:739)
                      at android.os.Handler.dispatchMessage(Handler.java:95)
                      at android.os.Looper.loop(Looper.java:135)
                      at android.app.ActivityThread.main(ActivityThread.java:5930)
                      at java.lang.reflect.Method.invoke(Native Method)
                      at java.lang.reflect.Method.invoke(Method.java:372)
                      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1405)
                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1200)
```

natario1, 2017-10-26:
No idea. Make sure you are using the latest version

Aissa-H, 2017-10-26:
I had a useless dependency to CameraKit lib in my gradle, by removing it, it's working (maybe some conflict of classes?) ! :+1:

natario1, 2017-10-26:
Yes, likely our class was loading their XML file.

closed 2017-10-26 by Aissa-H