natario1/CameraView #84: Swap facing makes camera freeze

ManueGE, 2017-10-31:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [x] *I am using the latest version*
- [x] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

##### Detailed steps to reproduce the issue:

1. Open app
2. press the toggle facing button

##### Expected behavior:

The camera should swap from back to front

##### Actual behavior:

The camera freezes.

##### Notes:
I am running demo app in a Nexus 5, Android 6.0.1

natario1, 2017-10-31:
I can reproduce this, it used to work fine, thanks for reporting.

closed 2017-10-31 by natario1

ManueGE, 2017-10-31:
Great, any estimated date for new release?

natario1, 2017-10-31:
Now

PR #86 (2017-10-31): Release PreviewCallback during stop events
Fixes #84