natario1/CameraView #87: Improve Frame recycling

natario1, 2017-10-31:
We could use `camera1.setPreviewCallbackWithBuffer` instead of `camera1.setPreviewCallback` for frame processors.

Right now, each preview frame allocates a new byte[] array, even if there are no registered frame processors.

andrewmunn, 2017-11-01:
You could require any registered frame processors to call a `finished()` method when they're done with the buffer.

natario1, 2017-11-01:
This was the original idea, there is `release()`, but there are two possible roads:

- if a frame comes and the processor is still managing the other, allocate a new array. This has a memory footprint and is harder to implement, but ensures that the processor won't lose any frame
- if a frame comes and the processor is still managing the other, consider it lost - we only have a single byte[] array cycling. This has no memory downsides and is extremely easy to implement, but the processor might not receive all frames

I don't know what's best at the moment

andrewmunn, 2017-11-01:
Before using `CameraView`, I used Google's sample code for using the barcode vision library. Checkout [CameraSource](https://github.com/googlesamples/android-vision/blob/master/visionSamples/barcode-reader/app/src/main/java/com/google/android/gms/samples/vision/barcodereader/ui/camera/CameraSource.java#L808).

In short, CameraSource adds 4 frame buffers, and then a background processing thread for barcode detection. While frame processing is ongoing, the byte[] array is not in the camera queue. It is released after it finishes. But because there are 4 buffers being cycled, there is always one available for the next frame.

As soon as the frame processing finishes, it begins work on the most recent frame.

 I've managed to replicate this behavior using `CameraView` and RxJava2. As new frames are delivered, I throw them into an RxJava `Flowable `that only keeps the latest frame, and then I do processing on yet another background thread. Once processing finishes, I request the latest frame from the Flowable. The downside here is I may be using more memory than before. It's not clear how many frame buffer's the Camera1 API allocates itself.

natario1, 2017-11-01:
This is pretty much Option 1, but I am concerned about the delay issue. Let's say that we can allocate infinite arrays, camera sends 1f each 20 ms (50 fps), processing a frame takes 100ms.

Even with an infinite queue, soon the processor won't be in sync with the preview, it will be processing old (very old) frames, and this gets worse.

I have never done frame processing, so I don't know if a frame processor prefers to have all frames but possibly delayed, or to have most of them but surely in sync. I will take a look at the project

andrewmunn, 2017-11-01:
For my usage, processing a frame actually takes about 40 ms, and each frame is 33ms (30 FPS). I don't want to fall behind in processing, so what I do is just skip frames if I can't keep up. That's how `CameraSource` works, because it just discards frames that come in during processing, by only retaining the latest frame.

natario1, 2017-11-01:
I see. Thanks, I'll think about this and set up a PR

PR #94 (2017-11-06): Improve frame processing - recycle buffers and Frame instances
Fixes #87 , should give a huge improvement in memory consumption and efficiency.

Both byte buffers and `Frame` instances are automatically recycled, with small impact on who's not using processors at all.
closed 2017-11-06 by natario1