natario1/CameraView #90: Activity Leak caused by HandlerThread

andrewmunn, 2017-11-01:
CameraView is so awesome. I just implemented the Frame processor for barcode detection. However, Leakcanary keeps telling me there is a memory leak!

- [x ] *I am using the latest version* (1.3.1)
- [ ] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. Install [Leakcanary](https://github.com/square/leakcanary)
2. Use CameraView in an Activity.
3. finish the Activity with  a CameraView and get alerted of the activity leaking.

#### Expected behavior:

Activity is not leaked.

#### Actual behavior:

It's leaked because the `HandlerThread` doesn't get garbage collected. You need to call the `quit()` method of the HandlerThread. [See this medium post.](https://medium.com/@ali.muzaffar/handlerthreads-and-why-you-should-be-using-them-in-your-android-apps-dc8bf1540341)

Or possibly you might need to set the `setUncaughtExceptionHandler` to `null`.

#### Interesting logs:
The leak trace.
```
 co.stackstorage.mover.feature.createitem.CreateItemActivity has leaked:
* GC ROOT android.os.HandlerThread.<Java Local>
* references android.os.HandlerThread.uncaughtExceptionHandler
* references com.otaliastudios.cameraview.CameraController$1.this$0 (anonymous implementation of java.lang.Thread$UncaughtExceptionHandler)
* references com.otaliastudios.cameraview.Camera1.mCameraCallbacks
* references com.otaliastudios.cameraview.CameraView$Callbacks.this$0
* references com.otaliastudios.cameraview.CameraView.mContext
* leaks co.stackstorage.mover.feature.createitem.CreateItemActivity instance
```

natario1, 2017-11-01:
Setting the uncaughtExceptionHandler to null makes a lot of sense. Can you try to do that in `CameraController.stopImmediately` and see if this is fixed?

andrewmunn, 2017-11-01:
Yep, looks like adding `mHandler.getThread().setUncaughtExceptionHandler(null);` into `stopImmediately` fixes the issue.

natario1, 2017-11-01:
Cool! Can you file a PR as well? Or I'll do it

andrewmunn, 2017-11-01:
I'll file a PR!

closed 2017-11-01 by natario1

orange7, 2018-05-19:
I am using the latest version (1.5.0)，The problem still exists。

commit f3317ad4 (2017-11-01): Remove UncaughtExceptionHandler from CameraController in stopImmediately (#92)

* Fixes #90 Remove UncaughtExceptionHandler from CameraController in stopImmediately

* Move setUncaughtExceptionHandler(null) before onStop()