natario1/CameraView #91: Image quality 100% takes a huge amount of memory

RobertoMorelos, 2017-11-01:
- [x] *I am using the latest version*
- [x] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. Set the quality image to 100.
2.Take a photo and display the bitmap in a image view.
3.Check the memory usage.

#### Expected behavior:
App should not consume a huge amount of memory.

#### Actual behavior:
App is consuming a huge amount of memory and can lead to an Out of Memory Exception.

natario1, 2017-11-01:
You can load a smaller version using CameraUtils.loadBitmap(byteArray, maxWidth, maxHeight). Uncompressed images *are* huge

RobertoMorelos, 2017-11-01:
I used the cameraCropOutput feature and the generated byte array was ok, but you first create a bitmap and crop it, that bitmap is consuming a huge amount of memory.

natario1, 2017-11-01:
It's true that we are missing a `recycle()`, do you want to create a PR? We should add `crop.recycle()` at the end of `CropHelper.cropToJpeg`.

RobertoMorelos, 2017-11-01:
Sure, let me check that!

RobertoMorelos, 2017-11-01:
We can use Glide for decoding byte arrays. The other way is to resample the image. I tried using the recycler pero there is no difference, pretty much the same.

RobertoMorelos, 2017-11-01:
Working on resampling the image. Found a pretty good improvement.

natario1, 2017-11-01:
Resampling (cropping) the byte[] array? Feel free to discuss here

RobertoMorelos, 2017-11-01:
No, I meant the bitmap (not sure if you can do that with a byte array).

SidhanshuUdawat, 2018-01-29:
@natario1 Do you guys need any help here ? Coz i've used the lib for my app and was facing the same OOM issue and eventually had to add the cropping logic at the library level.

natario1, 2018-01-29:
I was planning to remove this feature, honestly. What do you mean by adding the logic at the library level? @SidhanshuUdawat

wosiu, 2018-10-21:
> Apart from this, `cameraCropOutput` has this downside, we can't avoid loading the full bitmap in memory. Unless someone is able to crop the JPEG directly from the `byte[]` array

I believe BitmapRegionDecoder does that. It calculate start byte and size of gaps between blocks of bytes. To me it might be removed. It is so easy to handle this case on developer side like:
```
        Rect roi = new Rect(what you like to crop..);
        Bitmap inputBitmap = BitmapRegionDecoder.
                newInstance(frameBytes, 0, frameBytes.length, true).
                decodeRegion(roi, options);
```
and btw we can use some nice `options` for e.g.: sampling down, or getting exact width and heigh, or reuse prealocated bitmap..