natario1/CameraView #98: Possible race condition. Quick rotations result in crash.

Aberrational, 2017-11-14:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [x] *I am using the latest version*
- [x] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. Open demo app
2. Rotate device 
3. Rotata again (quickly!)
4. Loop 3. few times until crash (race condition?)

#### Expected behavior:
no crash

#### Actual behavior:
crash

#### Screenshots:

#### Interesting logs:
```

11-14 15:09:12.620 11511-11642/com.otaliastudios.cameraview.demo I/CameraController: Stop: executing. State: STATE_STARTED
11-14 15:09:12.620 11511-11642/com.otaliastudios.cameraview.demo I/CameraController: Stop: about to call onStop()
11-14 15:09:12.620 11511-11642/com.otaliastudios.cameraview.demo I/Camera1: onStop: About to clean up.
11-14 15:09:12.620 11511-11642/com.otaliastudios.cameraview.demo I/Camera1: onStop: Clean up. Ending video.
11-14 15:09:12.620 11511-11642/com.otaliastudios.cameraview.demo I/Camera1: endVideoImmediately: is capturing: false
11-14 15:09:12.620 11511-11642/com.otaliastudios.cameraview.demo I/Camera1: onStop: Clean up. Stopping preview.
11-14 15:09:12.620 11511-11511/com.otaliastudios.cameraview.demo I/CameraController: destroy: state: STATE_STOPPING
11-14 15:09:12.620 11511-11511/com.otaliastudios.cameraview.demo I/CameraController: stopImmediately: State was: STATE_STOPPING
11-14 15:09:12.620 11511-11511/com.otaliastudios.cameraview.demo I/Camera1: onStop: About to clean up.
11-14 15:09:12.620 11511-11511/com.otaliastudios.cameraview.demo I/Camera1: onStop: Clean up. Ending video.
11-14 15:09:12.620 11511-11511/com.otaliastudios.cameraview.demo I/Camera1: endVideoImmediately: is capturing: false
11-14 15:09:12.620 11511-11511/com.otaliastudios.cameraview.demo I/Camera1: onStop: Clean up. Stopping preview.
11-14 15:09:12.830 11511-11511/com.otaliastudios.cameraview.demo I/Camera1: onStop: Clean up. Stopped preview.
11-14 15:09:12.840 11511-11511/com.otaliastudios.cameraview.demo I/Camera1: onStop: Clean up. Releasing camera.
11-14 15:09:12.850 11511-11642/com.otaliastudios.cameraview.demo W/Camera1: onStop: Clean up. Exception while stopping preview. java.lang.RuntimeException: Camera is being used after Camera.release() was called
                                                                            java.lang.RuntimeException: Camera is being used after Camera.release() was called
                                                                                at android.hardware.Camera.setHasPreviewCallback(Native Method)
                                                                                at android.hardware.Camera.setPreviewCallbackWithBuffer(Camera.java:912)
                                                                                at com.otaliastudios.cameraview.Camera1.onStop(Camera1.java:217)
                                                                                at com.otaliastudios.cameraview.CameraController$4.run(CameraController.java:170)
                                                                                at android.os.Handler.handleCallback(Handler.java:739)
                                                                                at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                at android.os.Looper.loop(Looper.java:158)
                                                                                at android.os.HandlerThread.run(HandlerThread.java:61)
11-14 15:09:12.850 11511-11642/com.otaliastudios.cameraview.demo I/Camera1: onStop: Clean up. Releasing camera.
11-14 15:09:12.850 11511-11642/com.otaliastudios.cameraview.demo I/Camera1: onStop: Clean up. Released camera.
11-14 15:09:12.860 11511-11642/com.otaliastudios.cameraview.demo E/AndroidRuntime: FATAL EXCEPTION: CameraViewController
                                                                                   Process: com.otaliastudios.cameraview.demo, PID: 11511
                                                                                   com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Camera is being used after Camera.release() was called
                                                                                       at com.otaliastudios.cameraview.Camera1.onStop(Camera1.java:240)
                                                                                       at com.otaliastudios.cameraview.CameraController$4.run(CameraController.java:170)
                                                                                       at android.os.Handler.handleCallback(Handler.java:739)
                                                                                       at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                       at android.os.Looper.loop(Looper.java:158)
                                                                                       at android.os.HandlerThread.run(HandlerThread.java:61)
                                                                                    Caused by: java.lang.RuntimeException: Camera is being used after Camera.release() was called
                                                                                       at android.hardware.Camera.setHasPreviewCallback(Native Method)
                                                                                       at android.hardware.Camera.setPreviewCallbackWithBuffer(Camera.java:912)
                                                                                       at com.otaliastudios.cameraview.Camera1.onStop(Camera1.java:217)
                                                                                       at com.otaliastudios.cameraview.CameraController$4.run(CameraController.java:170) 
                                                                                       at android.os.Handler.handleCallback(Handler.java:739) 
                                                                                       at android.os.Handler.dispatchMessage(Handler.java:95) 
                                                                                       at android.os.Looper.loop(Looper.java:158) 
                                                                                       at android.os.HandlerThread.run(HandlerThread.java:61) 
11-14 15:09:13.030 11511-11511/com.otaliastudios.cameraview.demo I/Camera1: onStop: Clean up. Released camera.
```

*You can use `CameraLogger.setLogLevel(LEVEL_VERBOSE)`
to add logs to logcat, and `CameraLogger.registerLogger()`
to handle logs yourself (e.g. export to file or a crash reporting service)
so you can collect data from your users.*

natario1, 2017-11-14:
Are you using v. 1.3.2 or the latest `master`? This might be already fixed by #97

Aberrational, 2017-11-14:
Demo app: latest master 
My own code (same issue): 
compile 'com.otaliastudios:cameraview:1.3.2'

It's not easy. You have to want to crash it. Don't always wait for the rotation to complete. Change the frequency. I can get it to crash in about 15 turns.

natario1, 2017-11-14:
Ok I understand why. 

stop() and destroy() are called simultaneously.

destroy() exceptions are swallowed and that's good.
stop() exceptions are caught by the uncaughtExceptionHandler.

But destroy() sets the uncaughtExceptionHandler to null.

I'll fix this

natario1, 2017-11-14:
No I don't really understand. Sorry for the question, is the app really crashing or are you just seeing this error in the logs? This should crash a separate thread.

natario1, 2017-11-14:
You can try this branch and see if it's fixed https://github.com/natario1/CameraView/tree/resolution

I can't reproduce on my device.

closed 2017-11-14 by natario1

Aberrational, 2017-11-15:
Confirming that the RuntimeException no longer crashes the app. Both for demo from latest master and 1.4.0 used in code.
Sorry for no response. Different timezones :) Thanks for super quick fix.

natario1, 2017-11-15:
Thanks for reporting!

commit 4f5ced72 (2017-11-14): Fix #98
commit dcf5ef41 (2017-11-14): Option to choose the picture output size (#99)

* Add SizeSelectors and tests

* Implement SizeSelectors in CameraController

* XML attrs, improve AspectRatio cache

* AspectRatio tests

* Flip sizes before passing to selectors

* Add README info

* Fix preview sizing bug

* Nits

* Fix #98