natario1/CameraView #100: CameraListener Called Twice

MatyasK, 2017-11-15:
HI, 
So I have a weird issue or bug. I have cameraview with button, and I mapped the TAp gesture to capture a photo or you can click the button, whatever is clicked the clicklistener called two times. Even if I disable the button or gesture. 

Here is my layout:
`  <FrameLayout
        android:id="@+id/camera_frame"
        android:paddingTop="?attr/actionBarSize"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        app:layout_constraintEnd_toEndOf="parent"
        app:layout_constraintStart_toStartOf="parent"
        app:layout_constraintTop_toBottomOf="@+id/toolbar"
        app:layout_constraintVertical_bias="0.0"
       >

        <com.otaliastudios.cameraview.CameraView
            android:id="@+id/camera"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:keepScreenOn="true"
            app:cameraJpegQuality="10"
            app:cameraSessionType="picture">

        </com.otaliastudios.cameraview.CameraView>

        <Button
            android:id="@+id/captureColor"
            android:text="@string/scan_color"
            android:textSize="30sp"
            android:background="@color/md_white_1000"
            android:layout_width="match_parent"
            android:layout_height="80dp"
            android:layout_marginLeft="25dp"
            android:layout_marginRight="25dp"
            android:layout_marginBottom="15dp"
            />
    </FrameLayout>
` 
And here is my activity:
`
    @ViewById(R.id.camera)
    CameraView cameraView;

    @AfterViews
    void init() {
        setSupportActionBar(toolbar);
    

        cameraView.mapGesture(Gesture.TAP, GestureAction.CAPTURE);
        cameraView.addCameraListener(new CameraListener() {
            @Override
            public void onPictureTaken(byte[] picture) {
                Log.d(TAG, "onPictureTaken: called");
                CameraUtils.decodeBitmap(picture, new CameraUtils.BitmapCallback() {
                    @Override
                    public void onBitmapReady(Bitmap srcBmp) {
                     
                    }
                });
            }
        });
    }

    @Click(R.id.captureColor)
    void captureWithButton() {
        cameraView.capturePicture();

    }`

natario1, 2017-11-15:
Maybe your init method is called twice.

MatyasK, 2017-11-15:
totally my bad, in my base activity I called init method as well. You saved my day. Thank you :)

closed 2017-11-15 by natario1