natario1/CameraView #102: Issue in capture video

Sun1718, 2017-11-15:
Hi, 
First, thanks for your useful codes. I have tested demo app with Sumsung Galaxy Core (android 5.1.1). Everything is OK, but
1- captured video with back camera is not saved correctly.(screenshot is attached)
2- captured video with front camera is OK, but it is flipped horizontaly.

- [*] *I am using the latest version*
- [*] *I can reproduce the bug in the demo app, and if not...*
- [*] *I have tried to reproduce the bug in the demo app*

#### Screenshots:
[image]

#### Interesting logs:
[logcat.txt](https://github.com/natario1/CameraView/files/1475239/logcat.txt)

Regards.

Sun1718, 2017-11-15:
Thanks for your quick reply.

I've attached demo app logs:
[log.txt](https://github.com/natario1/CameraView/files/1475765/log.txt)

natario1, 2017-11-15:
What if you reproduce the video ( /data/data/com.otaliastudios.cameraview.demo/files/video.mp4 ) with another app, can you do that? You can edit the demo app and choose another directory if it helps.

I don't see anything strange in your logs, not sure I can help!

Sun1718, 2017-11-15:
I changed video path, recorded video is same as that screenshot.
Then I changed video quality from menu to higest and it recorded correctly.(Actual Resolution is 720*1280)

What is youe idea?

Thanks again.

natario1, 2017-11-15:
Uhm.. No idea!

Sun1718, 2017-11-18:
I've checked it. The result was same as 1.4.

natario1, 2017-11-18:
Then I really don't know why, sorry.

eirtaza, 2018-02-15:
I am facing the same problem with Samsung S4 Active.