natario1/CameraView #103: queueBuffer: BufferQueue has been abandoned

buelersandra, 2017-11-15:
how to prevent "queueBuffer: BufferQueue has been abandoned" when view is dismissed from screen with taking a picture after Start() is called.

natario1, 2017-11-15:
I don't know what that means

buelersandra, 2017-11-15:
[image]

natario1, 2017-11-19:
It's because we release stuff asynchronously (e.g. not exactly `onStop` but some moments later). See this very good answer: https://stackoverflow.com/a/22490648/4288782 .

We release the camera *moments after* the surface has been destroyed, and that's where you get that warning, until the camera is destroyed too. I think we can safely ignore it.

closed 2017-11-19 by natario1