natario1/CameraView #107: Any way to disable auto focus?

vedmedenko, 2017-11-19:
Hi! 

Is it possible to disable camera auto focus with your library?

Thank you in advance!

natario1, 2017-11-19:
No. May I ask why you want to disable it?

vedmedenko, 2017-11-19:
I want to calibrate android camera using OpenCV methods for later use. That process, as far as i know, requires the camera focus to be locked. Your great library seems to be the best choice since i want to focus on other stuff instead of dealing with the camera, however i am missing this feature.

natario1, 2017-11-19:
Calibrate what? (Just curious).

This feature was present but I have removed it months ago. I had (have) the feeling that 99.99% of users would just need the best focus option.

Also focus modes are complicated to wrap in a useful API. Maybe Focus.BEST and Focus.OFF. if anyone wants to work on it we can discuss.

Flashspeed, 2019-03-14:
> No. May I ask why you want to disable it?

Would be nice to disable it so it can be locked to Infinite focus mode

edward1988, 2019-04-08:
+1 May I ask, do you plan to add this functionality(disable auto focus)?

Geraldo-Neto, 2019-07-30:
Would really be great to have more focus modes. But the best we can do for now is to set autofocus reset delay to 0 then focus on a given point. Like this:
`camera.setAutoFocusResetDelay(0); camera.startAutoFocus(camera.getWidth()/2,camera.getHeight()/2);`

natario1, 2019-10-02:
Not planning to do this. But you can change focus in `Camera2` by simply subclassing the engine:

- Subclass `CameraView`, e.g. `MyCameraView`
- Subclass `Camera2Engine`, e.g. `MyCamera2Engine`
- Override `MyCameraView.instantiateCameraEngine()`, and return an instance of `My2CameraEngine`
- In `My2CameraEngine`, override protected methods like `applyDefaultFocus()` to do whatever you want

closed 2019-10-02 by natario1

wosiu, 2019-10-06:
Thank you for instructions :) Should be fine with it.