natario1/CameraView #108: Rotated preview on front camera

Monster-king, 2017-11-20:
*If you are trying to report a bug, please file the information below.
Prerequisites:*
Preview rotated to 180 degrees on front camera (only preview)

- [ x] *I am using the latest version*
- [x ] *I can reproduce the bug in the demo app, and if not...*
- [ x] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. Take Samsung Galaxy Tab S (android 6.0.1)
2. Open CameraView demo app
3. Change camera to front

#### Expected behavior:
Normal preview
#### Actual behavior:
Rotated preview to 180 degrees
#### Screenshots:
[image]

natario1, 2017-11-20:
Thanks. I don't have the device so we need more info.

1. Can you post logs coming from the demo app?

2. Can you try with another library, like this https://github.com/afollestad/material-camera ? What happens? Just to understand if it's a device specific issue.

Monster-king, 2017-11-21:
1. there logs:
[logs.txt](https://github.com/natario1/CameraView/files/1490248/logs.txt)
2. I try https://github.com/afollestad/material-camera, but it don't work on my device. Instead it I tried https://github.com/google/cameraview and there error not observed

natario1, 2017-11-21:
I think I have fixed this in #112 .

You can give it a try before we merge if you want. It's just one line to change by hand.

closed 2017-11-22 by natario1

Monster-king, 2017-11-23:
Working fine

PR #112 (2017-11-22): Change the offset computation for front cameras
This should fix #108 and possibly not break anything else.

The old formula was giving correct results for `sensorOffset = 90` and `sensorOffset = 270`, but not in the other cases (`0` and `180`). That might be the reason why no-one has been reporting it.