natario1/CameraView #109: Very slow cropping

Monster-king, 2017-11-20:
Taking picture without cropping output takes ~0.5 - 0.8 seconds. When set crop output to true it takes ~3 seconds. Is there any way to make it faster?

natario1, 2017-11-20:
It's not only slow, it's also prone to out of memory errors #91 . We take the JPEG, decompress to `Bitmap` in full resolution, crop the `Bitmap`, compress again to JPEG. This is a relic from CameraKit and it sucks.

- To make it faster in your app you can simply not use it. Use `wrap_content` for width and height so there is nothing to be cropped.

- To make it faster in the library we need to use the NDK and do the cropping in native code. There are native libraries ([example](http://www.ben.com/jpeg/)) that can crop a JPEG file without even decompressing, no middle steps. That would be super fast and no OOM at all.

If some of you wants to inject the NDK in `CameraView`, that's very welcome!

natario1, 2017-11-20:
This looks easy:

- clone https://github.com/libjpeg-turbo/libjpeg-turbo
- build a .so file for armeabi-v7a using [instructions](https://github.com/libjpeg-turbo/libjpeg-turbo/blob/master/BUILDING.md#armv7-32-bit-1)
- add the file to CameraView
- add their [java bindings](https://github.com/libjpeg-turbo/libjpeg-turbo/blob/master/java)

In this case we don't need the NDK at runtime (just for building the .so?) and we get bindings already?

xp-vit, 2018-02-04:
I don't think that this is very important and necessary. the reasons are following:
 - .so files will need to be included for all platforms, which results in significant increase of lib size
 - it is possible to fail more gracefully using WeakReferences for example
 - I'll take a closer look at current crop task in order to try optimizing current implementation

natario1, 2018-03-02:
I also don't think this is very important, at least it's not coming soon... let's close for now.

closed 2018-03-02 by natario1