natario1/CameraView #111: Lagging and blackened camera preview on accepting permissions on Android 6.0.1 or above

brn2393, 2017-11-21:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [ x] *I am using the latest version*
- [ x] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. add camera and storage permissions in manifest
2. deny permissions on Android 6.0.1 before starting camera activity
3. start activity and accept the permissions
4. camera preview starts with blackened screen and lag.

#### Expected behavior: activity needs to work smooth without blackened preview and without lag when permissions are accepted

#### Actual behavior: camera preview starts with lagging, activity is observed to behave strange (pausing, resuming, pausing again).

#### Screenshots: 

#### Interesting logs:
11-21 16:13:38.791 23790-23790/com.*.android I/CameraPreview: onSurfaceAvailable: w= 720 h= 1232
11-21 16:13:38.792 23790-23790/com.*.android I/Camera1: onSurfaceAvailable: Size is 720x1232
11-21 16:13:38.979 23790-23822/com.*.android D/FA: Connected to remote service
11-21 16:13:38.980 23790-23822/com.*.android V/FA: Processing queued up service tasks: 5
11-21 16:13:38.980 23790-23790/com.*.android W/CameraView: onMeasure: surface is not ready. Calling default behavior.
11-21 16:13:38.980 23790-23790/com.*.android W/CameraView: onMeasure: surface is not ready. Calling default behavior.
11-21 16:13:48.719 23790-23822/com.*.android V/FA: Using measurement service
11-21 16:13:48.719 23790-23822/com.*.android V/FA: Connecting to remote service
11-21 16:13:48.721 23790-23790/com.*.android I/CameraCallbacks: onDisplayOffsetChanged 0
11-21 16:13:48.721 23790-23790/com.*.android I/CameraController: Start: posting runnable. State: STATE_STOPPED
11-21 16:13:48.722 23790-23871/com.*.android I/CameraController: Start: executing. State: STATE_STOPPED
11-21 16:13:48.722 23790-23871/com.*.android I/CameraController: Start: about to call onStart() STATE_STARTING
11-21 16:13:48.727 23790-23822/com.*.android V/FA: Activity resumed, time: 179922419
11-21 16:13:48.733 23790-23822/com.*.android V/FA: Screen exposed for less than 1000 ms. Event not sent. time: 14
11-21 16:13:48.733 23790-23822/com.*.android V/FA: Using measurement service
11-21 16:13:48.733 23790-23822/com.*.android V/FA: Connection attempt already in progress
11-21 16:13:48.734 23790-23822/com.*.android V/FA: Activity paused, time: 179922433
11-21 16:13:48.836 23790-23822/com.*.android D/FA: Connected to remote service
11-21 16:13:48.836 23790-23822/com.*.android V/FA: Processing queued up service tasks: 2
11-21 16:13:48.907 23790-23822/com.*.android V/FA: Activity resumed, time: 179922607
11-21 16:13:48.908 23790-23790/com.*.android I/CameraCallbacks: onDisplayOffsetChanged 0
11-21 16:13:48.908 23790-23790/com.*.android I/CameraController: Start: posting runnable. State: STATE_STARTING
11-21 16:13:48.912 23790-23822/com.*.android V/FA: Screen exposed for less than 1000 ms. Event not sent. time: 5
11-21 16:13:48.913 23790-23822/com.*.android V/FA: Activity paused, time: 179922612
11-21 16:13:48.952 23790-23871/com.*.android I/Camera1: onStart: Applying default parameters.
11-21 16:13:48.972 23790-23871/com.*.android I/Camera1: bindToSurface: Started
11-21 16:13:48.978 23790-23871/com.*.android I/Camera1: size: sizesFromList: [2592x1944, 2592x1458, 2080x1560, 2080x1170, 2048x1536, 1920x1080, 1600x1200, 1600x900, 1280x720, 800x600, 720x480, 640x480, 352x288, 320x240, 176x144]
11-21 16:13:48.978 23790-23871/com.*.android I/Camera1: shouldFlipSizes: mDeviceOrientation= 0 mSensorOffset= 270
11-21 16:13:48.978 23790-23871/com.*.android I/Camera1: shouldFlipSizes: sensorToDisplay= 90
11-21 16:13:48.978 23790-23871/com.*.android I/CameraController: computePictureSize: flip: true
11-21 16:13:48.979 23790-23871/com.*.android I/CameraController: computePictureSize: result: 1944x2592
11-21 16:13:48.983 23790-23871/com.*.android I/Camera1: size: sizesFromList: [1920x1080, 1280x960, 1280x720, 800x600, 720x480, 640x480, 352x288, 320x240, 176x144]
11-21 16:13:48.983 23790-23871/com.*.android I/Camera1: shouldFlipSizes: mDeviceOrientation= 0 mSensorOffset= 270
11-21 16:13:48.983 23790-23871/com.*.android I/Camera1: shouldFlipSizes: sensorToDisplay= 90
11-21 16:13:48.984 23790-23871/com.*.android I/CameraController: size: computePreviewSize: targetRatio: 4:3 targetMinSize: 1232x720
11-21 16:13:48.985 23790-23871/com.*.android I/CameraController: computePreviewSize: result: 1280x960
11-21 16:13:48.985 23790-23871/com.*.android I/Camera1: bindToSurface: Dispatching onCameraPreviewSizeChanged.
11-21 16:13:48.985 23790-23871/com.*.android I/CameraCallbacks: onCameraPreviewSizeChanged
11-21 16:13:48.985 23790-23871/com.*.android I/Camera1: shouldFlipSizes: mDeviceOrientation= 0 mSensorOffset= 270
11-21 16:13:48.985 23790-23871/com.*.android I/Camera1: shouldFlipSizes: sensorToDisplay= 90
11-21 16:13:48.986 23790-23871/com.*.android I/CameraPreview: setDesiredSize: desiredW= 960 desiredH= 1280
11-21 16:13:48.999 23790-23871/com.*.android I/Camera1: bindToSurface: Starting preview with startPreview().
11-21 16:13:49.019 23790-23790/com.*.android I/CameraPreview: crop: applied scaleX= 1.2833334
11-21 16:13:49.019 23790-23790/com.*.android I/CameraPreview: crop: applied scaleY= 1.0
11-21 16:13:49.023 23790-23790/com.*.android I/Camera1: shouldFlipSizes: mDeviceOrientation= 0 mSensorOffset= 270
11-21 16:13:49.023 23790-23790/com.*.android I/Camera1: shouldFlipSizes: sensorToDisplay= 90
11-21 16:13:49.024 23790-23790/com.*.android I/CameraView: onMeasure: requested dimensions are (720[EXACTLY]x1232[EXACTLY])
11-21 16:13:49.024 23790-23790/com.*.android I/CameraView: onMeasure: previewSize is (960.0x1280.0)
11-21 16:13:49.024 23790-23790/com.*.android W/CameraView: onMeasure: both are MATCH_PARENT or fixed value. We adapt. This means CROP_CENTER. (720x1232)
[... 14 lines cut ...]
11-21 16:13:49.254 23790-23822/com.*.android V/FA: Activity paused, time: 179922953
11-21 16:13:49.297 23790-23871/com.*.android I/Camera1: bindToSurface: Started preview.
11-21 16:13:49.297 23790-23871/com.*.android I/Camera1: onStart: Ended
11-21 16:13:49.298 23790-23871/com.*.android I/CameraController: Start: returned from onStart(). Dispatching. STATE_STARTING
11-21 16:13:49.298 23790-23871/com.*.android I/CameraCallbacks: dispatchOnCameraOpened com.otaliastudios.cameraview.CameraOptions@738b694
11-21 16:13:49.298 23790-23871/com.*.android I/CameraController: Start: executing. State: STATE_STARTED
11-21 16:13:49.298 23790-23871/com.*.android I/CameraController: Start: executing. State: STATE_STARTED
11-21 16:13:49.298 23790-23871/com.*.android I/CameraController: Start: executing. State: STATE_STARTED
11-21 16:13:49.481 23790-23790/com.*.android I/CameraCallbacks: onDisplayOffsetChanged 0
11-21 16:13:49.481 23790-23790/com.*.android I/CameraController: Start: posting runnable. State: STATE_STARTED
11-21 16:13:49.481 23790-23822/com.*.android V/FA: Activity resumed, time: 179923178
11-21 16:13:49.482 23790-23871/com.*.android I/CameraController: Start: executing. State: STATE_STARTED
11-21 16:13:49.484 23790-23790/com.*.android I/CameraController: Stop: posting runnable. State: STATE_STARTED
11-21 16:13:49.484 23790-23822/com.*.android V/FA: Screen exposed for less than 1000 ms. Event not sent. time: 6
11-21 16:13:49.484 23790-23871/com.*.android I/CameraController: Stop: executing. State: STATE_STARTED
11-21 16:13:49.484 23790-23871/com.*.android I/CameraController: Stop: about to call onStop()
11-21 16:13:49.484 23790-23871/com.*.android I/Camera1: onStop: About to clean up.
11-21 16:13:49.485 23790-23871/com.*.android I/Camera1: onStop: Clean up. Ending video.
11-21 16:13:49.485 23790-23822/com.*.android V/FA: Activity paused, time: 179923184
11-21 16:13:49.485 23790-23871/com.*.android I/Camera1: endVideoImmediately: is capturing: false
11-21 16:13:49.485 23790-23871/com.*.android I/Camera1: onStop: Clean up. Stopping preview.
11-21 16:13:49.612 23790-23871/com.*.android I/Camera1: onStop: Clean up. Stopped preview.
11-21 16:13:49.612 23790-23871/com.*.android I/Camera1: onStop: Clean up. Releasing camera.
11-21 16:13:49.638 23790-23790/com.*.android I/CameraCallbacks: onDisplayOffsetChanged 0
11-21 16:13:49.638 23790-23822/com.*.android V/FA: Activity resumed, time: 179923336

natario1, 2017-11-21:
We can't show any preview before having camera permissions so... it must be black. And opening the camera takes time.

I don't know why your activity goes through onResume / onPause so often. Maybe every permission that you ask triggers a new dialog. Maybe there are other dialogs. I don't know.

If this is bothering you, don't call start and stop if you don't have permissions (e.g. manage permissions yourself).

```
    @Override
    public void onResume() {
        super.onResume();
        if (mHasPermissions) {
            camera.start();
        }
    }

    @Override
    public void onPause() {
        super.onPause();
        if (mHasPermissions) {
            camera.stop();
        }
    }
```

brn2393, 2017-11-21:
no sir its not completely black screen. here's a screenshot. I am trying your solution now.
[image]

brn2393, 2017-11-22:
your solution didn't work. Any update on the issue?

natario1, 2017-11-22:
No, I don't think we are doing anything wrong. It might be just a device specific issue with your auto exposure. You can google this, there are lots of reports about darkened camera preview, even in popular apps. But no solution.

brn2393, 2017-11-22:
I guess there's bug with devices running on 6.0.1 or above. Anyway thanks for quick reply and the great library. Hope to see Camera2 api complete integration soon.

Danm72, 2018-01-11:
I'm seeing this behaviour also. Bumping exposure to full makes the video just barely visible.

natario1, 2018-01-24:
I don't think this is our fault guys. Might be some manufacturer issue. Closing for now.

closed 2018-01-24 by natario1

Danm72, 2018-01-24:
I've tested this on a Pixel XL2, One Plus one, Samsung Galaxy S5.

If its a manufacturer issue then its a lot of manufacturers..

natario1, 2018-01-24:
Or a sensor manufacturer issue, or an Application issue, I don't know. But in no way the library is intentionally blackening the preview. There are many reports on the internet about this. But sure, if you find the root cause to be inside the library we can fix it.