natario1/CameraView #113: Thoughts for v2

natario1, 2017-11-22:
I would say that `v1.4.X` is a very stable release, and there's just minor refinements to do.
Major changes would require API signature changes as well so I am starting thinking about a `v2` release based on feedback and requests. This is how I see it.

|API|Action|
|----|-------|
|`Audio` enum|**Improve** it, not just `ON` and `OFF` but rather audio options to be passed to the video recorders.|
|`onPictureTaken(byte[])`|**Refactor** to `onPictureTaken(PictureResult)`|
|`onVideoTaken(File)`|**Refactor** to `onVideoTaken(VideoResult)`|
|`PictureResult`|**Add**. Candidate APIs: `isSnapshot()`, `getTimestamp()`, `getLocation()`, `getSize()`, `getJpeg()`, `toBitmap(maxWidth, maxHeight, callback)`.|
|`VideoResult`|**Add**. Candidate APIs: `isSnapshot()`, `getTimestamp()`, `getLocation()`, `getSize()`, `getFile()`.|
|`setJpegQuality`|**Remove**. It makes no sense, we get a jpeg already from the camera, we can't choose the quality. It is only used when recompressing after post-processing.|
|`setCropOutput`|**Remove**. See below.|
|`getPreviewSize`|**Change**: it should return the size correctly flipped in the view dimensions, and cropped if the preview is cropped (we currently return the camera1 preview size but it's useless.). We should also adapt it to the video size when the session type is video.|
|`getSnapshotSize`|**Remove**. It is equal to `getPreviewSize`|
|`getVideoSize`|**Add**. Return the size of video outputs taken with `takeVideo`.|
|`capturePicture`|**Rename** to `takePicture`. Never crop the output, use the internal picture sizes.|
|`captureSnapshot`|**Rename** to `takePictureSnapshot`. Takes picture from preview frames. Always crop the output to the visible part. Cropping is already native in `YuvImage`, but rotating is not. We need either a native jpeg rotator, or (better) a library that writes EXIF orientation tags to jpeg buffers.|
|`startCapturingVideo`|**Rename** to `takeVideo`.|
|`stopCapturingVideo`|**Rename** to `stopVideo`.|
|`takeVideoSnapshot`|**Add**. This should be the video equivalent of `takePictureSnapshot`. It should take videos by collecting preview frames, and crop them to the visible part. So if preview is 1:1, we get 1:1 videos. Then mix with audio. This will be **API18** min, see the [grafika](https://stackoverflow.com/a/32927878/4288782) example.|
|`take*` APIs in short:|We would have `takePicture`, `takeVideo`, `takePictureSnapshot` and `takeVideoSnapshot`. The snapshot version will implicitly crop the output to the visible part, and will be lower quality of course.| 
|`VideoRecorder`|**Create**. Since we must record videos in two ways, create an abstract `VideoRecorder` class. One implementation will contain `MediaRecorder` stuff that is already working well. The other should contain `MediaCodec` stuff to create videos from frames. Of course, `takeVideo` uses the first while `takeVideoSnapshot` uses the second.|
|`Gesture.DOUBLE_TAP`|**Add**. This is not impossible, add it to the set of supported gestures.|

Feel free to discuss. No hurry, I expect this in the middle of 2018.

ychescale9, 2017-11-25:
Any timeline for supporting Camera2? Thanks.

natario1, 2017-11-25:
I see no urgency for that, so it will come *after* v2 unless someone works on that.

tomislavr, 2018-01-11:
I am using this library for media capture in a application which has encrypted media gallery and encrypts picture on saving it to the file. It would be perfect if possible to get video "stream" directly from your library so we can encrypt it before it gets saved in the file. Is it possible to have `takeVideo(OutputStream)` call, in any format?

Patte1808, 2018-10-18:
Hey, I noticed that you are adding the label `v2-fixed`.

For my usecase, I have to set the `orientation hint` to a specific value for video, but there seems to be no public API for that. Therefore, I currently have to maintain my own fork.

Will v2 contain more public APIs to set the values? 
By the way great work and library. Really love this library. :)