natario1/CameraView #116: Image aspect ratio does not fit view bounds in pre-Lollipop device (suspected)

lsw19921030, 2017-11-27:
(I tested in Android 4.4.1, 5.0.2, 7.0.0 and 7.1.1, only reproduced in 4.4.1. (API19)).

```
            app:cameraCropOutput="true"
```
The above line has been set in XML.
However, the preview image right after clicking shutter, is still distorted in an Android 4.4.1 device.
The image is horizontally squeezed. i.e. Things look thinner than in real world.

lsw19921030, 2017-11-27:
By the way, the output bitmap is correct. This aspect ratio problem appears only in the preview.

lsw19921030, 2017-11-28:
I have tried all kinds of attributes:
```
            android:layout_width="wrap_content"
            android:layout_height="wrap_content"
```
```
            android:layout_width="match_parent"
            android:layout_height="match_parent"
```
With or without 
```
            app:cameraCropOutput="true"
```
With or without
```
            app:cameraPictureSizeAspectRatio="4:3"
```

But none of them produce correct result.

I have also checked that hardware acceleration is turned on.

lsw19921030, 2017-11-28:
Here are the screenshots.

The "normal" preview (no picture token):
[image]

The distorted preview (The static preview when a picture is token):
[image]

And the real bitmap that I saved in the device:
[image]

As you can see, the real bitmap (saved in the device) is actually larger (contains area that is not shown in the preview). 
The distorted preview showed the horizontal larger area, but not the vertical larger area, and thus distorted.

natario1, 2017-12-03:
Hi, thanks for the report. What is the "static preview when a picture is taken"? 

We only have a preview (not distorted, I hope) and an output picture (not distorted, I hope). If I didn't get this, can you post a video?

lsw19921030, 2017-12-04:
You got it, it is the preview that you mentioned. I call it static because in my mind there are 2 previews:
(1) Continuous picture showing in SurfaceView, when user takes no action
(2) The preview that freeze for about 1 second, act as the preview for the picture that user has taken

And I mean (2) is this issue

natario1, 2018-02-27:
@siralam let me see if I got this, you mean that after you tap "take picture", the preview is distorted for some time (before loading the picture)?

siralam, 2018-02-27:
I don't think this is that hard to understand...
So there indeed are 2 kinds of previews, as I have mentioned:
(1) The preview of the **camera**, which will change and move when you move your device, **before tapping "take picture"**
(2) The preview of the **picture taken**, which will **not** change and move when your move your device, and is **after tapping "take picture"**. And when this preview is ended, CameraView returns to preview of (1).

And the distortion occurs in (2).  
I hope this should be clear enough?!

natario1, 2018-02-27:
2 is just an artifact, it's not something we can control, so I don't think this can be fixed. But I'll take a look.

(It should not happen if you use wrap_content for both dimensions and nothing else, but I don't know for sure)

natario1, 2018-02-27:
(if this is so easy to understand to you, maybe it's easy for you to fix it too?)

siralam, 2018-02-27:
At that time when I am choosing different libraries, only your library has this problem, no matter it is using Camera1 API or Camera2 API. So I think that should not be something that you cannot control, especially you mentioned in the doc that you have your own way to ensure previews are not distorted no matter what size of CameraView we specify (Which indicates you did something to the image returned from camera, before displaying it as a preview).

Understanding this issue and fixing it is of course 2 completely different difficulties, but I guess you are not saying this seriously, but rather offensively. Although I am not, I apologize if I made you feel offensive.

natario1, 2018-03-01:
No problem, I should have added some smile face.

I can not reproduce this on my devices. But the thing is that we do not change the preview after 'capture' is clicked, it's something done by the internals in your specific device. 

Is your device/app hardware accelerated? What does `cameraView.isHardwareAccelerated()` return?

closed 2019-01-04 by natario1