natario1/CameraView #117: Changing session is slow also previews are not the same

Monster-king, 2017-11-28:
- [x] *I am using the latest version*
- [x] *I can reproduce the bug in the demo app, and if not...*
- [x] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. Change session type from picture to video or from video to picture.

#### Expected behavior:
Fast changing and same previews
#### Actual behavior:
Changing session taking about 1.5 - 2 seconds, also preview in picture session is zoomed when preview in video session is not (see screenshots).
#### Screenshots:
when session type is video
[image]
when session type is photo
[image]

natario1, 2017-12-03:
This is all working as intended. We change the preview size when recording video, to match the video output ratio. The whole logic is a bit messy but that's how it is at the moment.

For this reason (close preview, change size, open preview) changing session type is slow. This is common in camera apps though.

Monster-king, 2017-12-04:
@natario1 How I can avoid this? Because in my case preview size not changing when session changes.

natario1, 2017-12-04:
You can just use video mode without switching.

Monster-king, 2017-12-04:
@natario1 Is it possible capture picture in video mode?

natario1, 2017-12-04:
Yes, but the preview size will try to match the *video* output size, not the *picture* output size. Might not be a big deal.

Monster-king, 2017-12-04:
@natario1 thx, works fine

closed 2018-02-27 by natario1