natario1/CameraView #118: Camera Preview Stretches

devoma, 2017-12-04:
When I start to record the video is stretched and after recording it becomes correct.

If the video quality is set to Lowest then  it is stretched vertically, that is, by height. And when set quality to 480p then it is stretched by witdth.

Maybe this information will help to understand the problem

Device: LG K8. Android 6.0.
I Am using the latest version.

natario1, 2017-12-04:
Does It happen with 1.4.0?
Please add logs from 1.4.1.

devoma, 2017-12-04:
**Lowest Quality logs:**

> 12-05 00:45:02.252 19177-19177/com.otaliastudios.cameraview.demo D/=========: onRecordStart
> 12-05 00:45:02.252 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: constructor
> 12-05 00:45:02.253 19177-19329/com.otaliastudios.cameraview.demo D/MediaRecorder: mAudioZoomEnable = 0
> 12-05 00:45:02.253 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: doCleanUp
> 12-05 00:45:02.253 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: setListener
> 12-05 00:45:02.253 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: setClientName
> 12-05 00:45:02.254 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: setCamera(0xb964f580,0xb9c8d9f0)
> 12-05 00:45:02.254 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: setVideoSource(1)
> 12-05 00:45:02.254 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: Call init() since the media recorder is not initialized yet
> 12-05 00:45:02.254 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: init
> 12-05 00:45:02.256 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: setAudioSource(0)
> 12-05 00:45:02.257 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: setOutputFormat(1)
> 12-05 00:45:02.257 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: setVideoFrameRate(15)
> 12-05 00:45:02.257 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: setVideoSize(176, 144)
> 12-05 00:45:02.257 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: setVideoEncoder(2)
> 12-05 00:45:02.258 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: setParameters(video-param-encoding-bitrate=192000)
> 12-05 00:45:02.258 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: setParameters(audio-param-number-of-channels=1)
> 12-05 00:45:02.258 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: setParameters(audio-param-sampling-rate=8000)
> 12-05 00:45:02.258 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: setAudioEncoder(1)
> 12-05 00:45:02.259 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: setParameters(audio-param-encoding-bitrate=12200)
> 12-05 00:45:02.259 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: setParameters(video-param-rotation-angle-degrees=90)
> 12-05 00:45:02.260 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: setOutputFile(56, 0, 0)
> 12-05 00:45:02.260 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: prepare
> 12-05 00:45:02.261 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: start
> 12-05 00:45:02.284 19177-19355/com.otaliastudios.cameraview.demo D/mali_winsys: new_window_surface returns 0x3000
> 12-05 00:45:02.367 19177-19190/com.otaliastudios.cameraview.demo D/AudioSystem: getIoDescriptor: ioHandle = 7832, index = -2, mIoDescriptors = 0xb9a4ab70
> 12-05 00:45:02.367 19177-19190/com.otaliastudios.cameraview.demo V/AudioSystem: ioConfigChanged: [Update mIoDescriptors] add ioHandle = 7832 -> descriptor = 0xa0c4977c
> 12-05 00:45:02.367 19177-19190/com.otaliastudios.cameraview.demo V/AudioSystem: ioConfigChanged() new input opened 7832 samplingRate 8000, format 0x1 channel mask 0x10 frameCount 160 deviceId 0
> 12-05 00:45:02.472 19177-19191/com.otaliastudios.cameraview.demo D/AudioSystem: getIoDescriptor: ioHandle = 7832, index = 1, mIoDescriptors = 0xb9a4ab70
> 12-05 00:45:02.472 19177-19191/com.otaliastudios.cameraview.demo V/AudioSystem: ioConfigChanged: [Update mIoDescriptors] replace ioHandle = 7832 -> descriptor = 0xa0b4a77c
> 12-05 00:45:02.472 19177-19191/com.otaliastudios.cameraview.demo V/AudioSystem: ioConfigChanged() new config for input 7832 samplingRate 8000, format 0x1 channel mask 0x10 frameCount 160 deviceId 4
> 12-05 00:45:02.679 19177-19329/com.otaliastudios.cameraview.demo E/Camera: Unknown message type -2147483648
> 12-05 00:45:03.298 19177-19329/com.otaliastudios.cameraview.demo E/Camera: Unknown message type -2147483648
> 12-05 00:45:03.435 19177-19329/com.otaliastudios.cameraview.demo E/Camera: Unknown message type -2147483648
> 12-05 00:45:04.734 19177-19329/com.otaliastudios.cameraview.demo E/Camera: Unknown message type -2147483648
> 12-05 00:45:05.488 19177-19329/com.otaliastudios.cameraview.demo E/Camera: Unknown message type -2147483648
> 12-05 00:45:05.557 19177-19329/com.otaliastudios.cameraview.demo E/Camera: Unknown message type -2147483648
> 12-05 00:45:10.260 19177-19329/com.otaliastudios.cameraview.demo I/Camera1: endVideoImmediately: is capturing: true
> 12-05 00:45:10.260 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: stop
> 12-05 00:45:10.263 19177-19220/com.otaliastudios.cameraview.demo V/MediaRecorder: message received msg=101, ext1=268436456, ext2=-1011
> 12-05 00:45:10.263 19177-19220/com.otaliastudios.cameraview.demo V/MediaRecorder: callback application
> 12-05 00:45:10.263 19177-19220/com.otaliastudios.cameraview.demo V/MediaRecorder: back from callback
> 12-05 00:45:10.301 19177-19191/com.otaliastudios.cameraview.demo V/MediaRecorder: message received msg=101, ext1=536871912, ext2=-1011
> 12-05 00:45:10.301 19177-19191/com.otaliastudios.cameraview.demo V/MediaRecorder: callback application
> 12-05 00:45:10.301 19177-19191/com.otaliastudios.cameraview.demo V/MediaRecorder: back from callback
> 12-05 00:45:10.341 19177-19329/com.otaliastudios.cameraview.demo V/MediaRecorder: doCleanUp
[... cut ...]
> 12-05 00:46:57.683 19177-19191/com.otaliastudios.cameraview.demo D/AudioSystem: getIoDescriptor: ioHandle = 7840, index = 1, mIoDescriptors = 0xb9a4ab70
> 12-05 00:46:57.683 19177-19191/com.otaliastudios.cameraview.demo V/AudioSystem: ioConfigChanged() input 7840 closed
> 12-05 00:46:57.683 19177-19191/com.otaliastudios.cameraview.demo V/AudioSystem: ioConfigChanged: [Update mIoDescriptors] remove ioHandle = 7840
> 12-05 00:46:57.684 19177-19220/com.otaliastudios.cameraview.demo V/MediaRecorder: destructor
> 12-05 00:46:57.692 19177-19177/com.otaliastudios.cameraview.demo D/=========: onRecordFinished

Monster-king, 2017-12-06:
@natario1 I've same issue too. #117 in this bug you can see screens and logs.

natario1, 2017-12-06:
Stretching means deformed. @devoma please answer my question about 1.4.0 and post a screenshot where the deformation is visible. Or better a video! Thanks 🙂 video quality is not important.

devoma, 2017-12-06:
@natario1, no, It happens with 1.4.1. I posted the video, check it.

Monster-king, 2017-12-27:
@natario1 any news about this issue?

natario1, 2018-01-16:
Nope, I'll be away from keyboard until the end of February. Feel free to discuss.

alenseferovic1, 2018-04-11:
I resolve this issue in a way that I call startCapturingVideo and stopCapturingVideo in the same time, after mCameraView.start() method in onResume(). Like this:
 mCameraView.start();
 File file = new File(getFilesDir(), "preview");
  mCameraView.startCapturingVideo(file);
  mCameraView.stopCapturingVideo();

closed 2019-01-04 by natario1