natario1/CameraView #119: Leaked 2 GPU objects! Fatal signal 6 (SIGABRT)

Monster-king, 2017-12-06:
- [x] *I am using the latest version*
- [*] *I can reproduce the bug in the demo app, and if not...*
- [] *I have tried to reproduce the bug in the demo app*
This issue observed on A3 after updating OS version to 7.0.0
#### Detailed steps to reproduce the issue:

1. Add cameraview inside fragment.
2. In activity add viewpager and add fake fragment and fragment with cameraview to the viewpager 
3. Open activity and hide it.

#### Expected behavior:
Normal hiding and without crashing
#### Actual behavior:
App crashing
#### Interesting logs:
Texture is using 0.00B, count = 1
OffscreenBuffer is using 0.00B, count = 0
Layer is using 0.00B, count = 1
GPU Memory Leak happened by Application!!(2 Objects) 
Below Textures or Layers was not cleared even though Application being closed.(It should be fixed by Application side)
[Leak]Texture size is 0, 720 x 720
[Leak]Layer size is 0, 720 x 720
Leaked 2 GPU objects!
*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
Build fingerprint: 'samsung/a3xeltexx/a3xelte:7.0/NRD90M/A310FXXU3CQI4:user/release-keys'
Revision: '2'
ABI: 'arm'
pid: 32514, tid: 32751, name: RenderThread  >>> xxx.xxx.app <<<
signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
Abort message: 'Leaked 2 GPU objects!'
    r0 00000000  r1 00007fef  r2 00000006  r3 00000008
    r4 ccc7f978  r5 00000006  r6 ccc7f920  r7 0000010c
    r8 eeaeb00c  r9 00000000  sl ee5e4764  fp ee5e4760
    ip 00000000  sp ccc7f2f8  lr ef3904c7  pc ef392d30  cpsr 60070010
backtrace:
    #00 pc 0004ad30  /system/lib/libc.so (tgkill+12)
    #01 pc 000484c3  /system/lib/libc.so (pthread_kill+34)
    #02 pc 0001dd99  /system/lib/libc.so (raise+10)
    #03 pc 00019521  /system/lib/libc.so (__libc_android_abort+34)
    #04 pc 00017160  /system/lib/libc.so (abort+4)
    #05 pc 0000c6bb  /system/lib/libcutils.so (__android_log_assert+114)
    #06 pc 0003cf75  /system/lib/libhwui.so
    #07 pc 000271e3  /system/lib/libhwui.so
    #08 pc 000257c1  /system/lib/libhwui.so
    #09 pc 00028231  /system/lib/libhwui.so
    #10 pc 00028ad3  /system/lib/libhwui.so
    #11 pc 00029b01  /system/lib/libhwui.so (_ZN7android10uirenderer12renderthread12RenderThread10threadLoopEv+80)
    #12 pc 0000e329  /system/lib/libutils.so (_ZN7android6Thread11_threadLoopEPv+144)
    #13 pc 0006a56d  /system/lib/libandroid_runtime.so (_ZN7android14AndroidRuntime15javaThreadShellEPv+80)
    #14 pc 00047f93  /system/lib/libc.so (_ZL15__pthread_startPv+22)
    #15 pc 0001a161  /system/lib/libc.so (__start_thread+6)

[bug (2).txt](https://github.com/natario1/CameraView/files/1535349/bug.2.txt)

Monster-king, 2017-12-11:
@natario1 What can you say about this issue?

natario1, 2017-12-11:
I don't know, looks like you are not calling start and stop properly from the view pager adapter. But I have no time to checkout viewpagers now.. sorry!

Monster-king, 2017-12-14:
@natario1 I made some changes in CameraView
```
public void destroy() {
        clearCameraListeners();
        clearFrameProcessors();
        mCameraController.destroy();
        if(mCameraPreview != null) {
            mCameraPreview.destroy();
            mCameraPreview = null;
        }
    }
```
In TextureCameraPreview and SurfaceCameraPreview
```
@Override
    public void destroy() {
        if(getView().getSurfaceTexture() != null) {
            ((FrameLayout)getView().getParent()).removeAllViews();
            mSurface = null;
        }
    }
```
And problem solved.

closed 2018-02-27 by natario1

elhamshahidi72, 2018-08-04:
I am getting such an error for map ,could you please tell what changes did u do? what was your code before u solve it?

usmanrana07, 2019-04-04:
I'm getting same issue in CameraView.