natario1/CameraView #122: startPreview failed

ritesh94, 2017-12-11:
E/CameraController: uncaughtException: Unexpected exception: java.lang.RuntimeException: startPreview failed
                                                                               java.lang.RuntimeException: startPreview failed
                                                                                   at android.hardware.Camera.startPreview(Native Method)
                                                                                   at com.otaliastudios.cameraview.Camera1.applySizesAndStartPreview(Camera1.java:156)
                                                                                   at com.otaliastudios.cameraview.Camera1.bindToSurface(Camera1.java:130)
                                                                                   at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:186)
                                                                                   at com.otaliastudios.cameraview.CameraController$3.run(CameraController.java:175)
                                                                                   at android.os.Handler.handleCallback(Handler.java:746)
                                                                                   at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                   at android.os.Looper.loop(Looper.java:148)
                                                                                   at android.os.HandlerThread.run(HandlerThread.java:61)

iftekhar-ahmed, 2017-12-11:
I get the same issue from my app's production users running android 6.0 (samsung & motorola devices) so far.

com.otaliastudios:cameraview:1.4.1

Fatal Exception: java.lang.RuntimeException: startPreview failed
       at android.hardware.Camera.startPreview(Camera.java)
       at com.otaliastudios.cameraview.Camera1.applySizesAndStartPreview(Camera1.java:156)
       at com.otaliastudios.cameraview.Camera1.bindToSurface(Camera1.java:130)
       at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:186)
       at com.otaliastudios.cameraview.CameraController$3.run(CameraController.java:175)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:148)
       at android.os.HandlerThread.run(HandlerThread.java:61)

natario1, 2017-12-11:
Does this happen with 1.4.0?

ritesh94, 2017-12-11:
It happens with 'com.otaliastudios:cameraview:1.4.1'

iftekhar-ahmed, 2017-12-11:
@natario1 I cannot confirm `1.4.0` is problematic as i used the `1.4.1` in production. However while development i think tried `1.4.0` without any problem. ITC I'd say the problem could well be in both versions. For your help, here are devices and android version details for problems reported so far in my app -

All problems in Android 6.0

Devices - Samsung Galaxy S5 & Galaxy A5, Motorola G Turbo Edition, Motorola G4 Play

natario1, 2017-12-11:
I have no idea what could cause this. The only thing might be some things that were introduced in 1.4.1. I recommend going back to 1.4.0 and try for now, sorry. Let me know if you have more info.

ritesh94, 2017-12-12:
ohk... Any idea when new version will be releasing ?

chekashovsoftermii, 2017-12-15:
I have same problem. 
Device: Xiaomi Redmi 3S.
OS: Android 6.0.1

natario1, 2017-12-15:
@checkashovsoftermii does it happen with 1.4.0?

chekashovsoftermii, 2017-12-15:
@natario1 Yes

natario1, 2018-02-27:
Can you add steps to reproduce the issue?

semihozkoroglu, 2018-03-27:
hello, i added my crash stacktrace. @natario1 could you help me? lib version 1.4.1

Fatal Exception: java.lang.RuntimeExceptionstartPreview failed Raw Text
--
  | android.hardware.Camera.startPreview (Camera.java)
  | com.otaliastudios.cameraview.Camera1.applySizesAndStartPreview (Camera1.java:156)
  | com.otaliastudios.cameraview.Camera1.bindToSurface (Camera1.java:130)
  | com.otaliastudios.cameraview.Camera1.onStart (Camera1.java:186)
  | com.otaliastudios.cameraview.CameraController$3.run (CameraController.java:175)
  | android.os.Handler.handleCallback (Handler.java:739)
  | android.os.HandlerThread.run (HandlerThread.java:61)

BrantK, 2018-04-24:
Hello, I am still seeing this issue on 1.5.0. It only happens on my S8 and Note 8 devices. I did figure out a fix in case anyone is having the same problem as me.

Cause: The cameraView isn't completely destroyed when loading to a new fragment, in my case a photo preview fragment.

Fix: To fix this I changed cameraView.stop() in my onPause() method to cameraView.destroy(). 

This is probably not intended but it works for now.

kumarkumar217, 2018-05-07:
Android Version Number: 5.0 
Model: SM-G900F 
Manufacturer: samsung 

java.lang.RuntimeException: startPreview failed
at android.hardware.Camera.startPreview(Native Method)
at com.otaliastudios.cameraview.Camera1.applySizesAndStartPreview(Camera1.java:160)
at com.otaliastudios.cameraview.Camera1.bindToSurface(Camera1.java:134)
at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:191)
at com.otaliastudios.cameraview.CameraController$3.run(CameraController.java:180)

Seachal, 2018-05-15:
Hello, I am still  have this startPreview failed on 1.5.0,   The case of a bug：
CameraActivity：startActivityForResult    intent(CameraActivity.this,  B.class    )
                              onActivityResult           intent(CameraActivity.this,  C.class)
CameraActivity -> B->CameraActivity ->C      ->CameraActivity-B-CameraActivity-C happen 
 RuntimeException startPreview failed.

wider2, 2018-05-25:
I have confirm the same exception on my Lenovo Api 23

05-24 15:23:10.932 11034-11034/com.cityappventure E/UncaughtException: java.lang.RuntimeException: setParameters failed
        at android.hardware.Camera.native_setParameters(Native Method)
        at android.hardware.Camera.setParameters(Camera.java:1953)
        at com.otaliastudios.cameraview.Camera1$19.run(Camera1.java:806)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:148)
        at android.os.HandlerThread.run(HandlerThread.java:61)

thib-rdr, 2018-06-04:
Hi, same issue with a Samsung Galaxy Tab S2

`E/CameraController: uncaughtException: Unexpected exception: java.lang.RuntimeException: startPreview failed
    java.lang.RuntimeException: startPreview failed
        at android.hardware.Camera.startPreview(Native Method)
        at com.otaliastudios.cameraview.Camera1$12$2.onPictureTaken(Camera1.java:521)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1115)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:154)
        at android.os.HandlerThread.run(HandlerThread.java:61)`

terlono, 2018-06-05:
Here's what i found. I downloaded the lib locally and attached it and after some mingling i discovered that if i put a catch on the exception generating lines:

`        try
        {
            mCamera.startPreview();
        }
        catch (Exception e)
        {
            Log.e(TAG, e.getMessage());
        }
`

Even if the exception happens the camera returns to normal. Also one thing, it happend only on devices which were using SurfaceTexture instead of SurfaceHolder. Probably something with setting the max height and width on the preview there.

natario1, 2018-06-05:
@terlono thanks for investigating.

Are you sure it's not the opposite? I would expect that this happens with `SurfaceHolder` (device / app with no hardware acceleration), not with `SurfaceTexture`.

natario1, 2018-06-05:
Can you reproduce this with the demo app? If so, can you post logs coming from it in a txt file? From start up to the moment it crashes.

terlono, 2018-06-05:
Sure thing, but that will have to wait untill evening since im on a commute right now.

natario1, 2018-06-05:
Thanks 👍 I'd be happy to fix this with your collaboration.

daniele93dc, 2018-06-09:
I have same error, library version 1.5.1, android version 6.0.1, device Galaxy s5

k-farhan, 2018-07-10:
Guys, do we've any workaround for the issue?

wider2, 2018-07-20:
yes, I have the same problem with 1.5.1
on Lenovo 6

AndroidRuntime: FATAL EXCEPTION: main
    Process: com.cityappventure, PID: 14385
    java.lang.RuntimeException: startPreview failed
        at android.hardware.Camera.startPreview(Native Method)
        at com.otaliastudios.cameraview.Camera1$12$2.onPictureTaken(Camera1.java:521)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1132)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:148)
        at android.os.HandlerThread.run(HandlerThread.java:61)
07-20 11:18:23.123 860-7028/? W/ActivityManager:   Force finishing activity

the-dagger, 2018-08-08:
Same issue here with  Galaxy Tab A 7.0, Android version 5.1.1, Lollipop.
Library version is 1.5.1.

satishverma30, 2018-08-27:
Facing same issue, if fixed please let me know

natario1, 2018-09-14:
With #265 you can now catch this exception in the error handler, instead of having the app crash.

I still think it would have been cool to understand if there is a bug in the library or not - and this will be more difficult with #265 because we'll get less reports. But we were not able to fix this so...

I will close this now. If you get this exception again (after #265 is merged), please open a new issue with CameraLogger logs so we can see if there's something wrong on our side.

closed 2018-09-14 by natario1

PriyaSindkar, 2018-10-04:
Still facing same issue with Moto G4 Plus, API-24. hardwardAccelerated is true and therefore app is using SurfaceCameraPreview

mayurpateltristate, 2019-11-20:
When i launch app i am facing this kind of exception

E/CameraEngine: uncaughtException: Unexpected exception: java.lang.RuntimeException: Unknown CameraPreview output class.
    java.lang.RuntimeException: Unknown CameraPreview output class.
        at com.otaliastudios.cameraview.engine.Camera1Engine.onStartBind(Camera1Engine.java:181)
        at com.otaliastudios.cameraview.engine.CameraEngine$7.call(CameraEngine.java:444)
        at com.otaliastudios.cameraview.engine.CameraEngine$7.call(CameraEngine.java:441)
        at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:117)
        at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:112)
        at com.google.android.gms.tasks.zzf.run(Unknown Source:2)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:129)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.execute(WorkerHandler.java:104)
        at com.google.android.gms.tasks.zze.onComplete(Unknown Source:6)
        at com.google.android.gms.tasks.zzr.zza(Unknown Source:19)
        at com.google.android.gms.tasks.zzu.zze(Unknown Source:132)
        at com.google.android.gms.tasks.zzu.continueWithTask(Unknown Source:74)
        at com.otaliastudios.cameraview.engine.Step.doStart(Step.java:112)
        at com.otaliastudios.cameraview.engine.Step.doStart(Step.java:105)
        at com.otaliastudios.cameraview.engine.CameraEngine.startBind(CameraEngine.java:441)
        at com.otaliastudios.cameraview.engine.CameraEngine.access$500(CameraEngine.java:123)
        at com.otaliastudios.cameraview.engine.CameraEngine$17$1$2.then(CameraEngine.java:735)
        at com.otaliastudios.cameraview.engine.CameraEngine$17$1$2.then(CameraEngine.java:730)
        at com.google.android.gms.tasks.zzp.run(Unknown Source:2)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:129)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.execute(WorkerHandler.java:104)
        at com.google.android.gms.tasks.zzo.onComplete(Unknown Source:6)
        at com.google.android.gms.tasks.zzr.zza(Unknown Source:19)
        at com.google.android.gms.tasks.zzu.zze(Unknown Source:132)
        at com.google.android.gms.tasks.zzu.onSuccessTask(Unknown Source:78)
        at com.otaliastudios.cameraview.engine.CameraEngine$17$1.call(CameraEngine.java:730)
        at com.otaliastudios.cameraview.engine.CameraEngine$17$1.call(CameraEngine.java:721)
        at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:117)
        at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:112)
        at com.google.android.gms.tasks.zzf.run(Unknown Source:2)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:129)
        at com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.execute(WorkerHandler.java:104)
        at com.google.android.gms.tasks.zze.onComplete(Unknown Source:6)
        at com.google.android.gms.tasks.zzr.zza(Unknown Source:19)
        at com.google.android.gms.tasks.zzu.zze(Unknown Source:132)
        at com.google.android.gms.tasks.zzu.continueWithTask(Unknown Source:74)
        at com.otaliastudios.cameraview.engine.Step.doStart(Step.java:112)
        at com.otaliastudios.cameraview.engine.Step.doStart(Step.java:105)
        at com.otaliastudios.cameraview.engine.CameraEngine$17.run(CameraEngine.java:721)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.os.HandlerThread.run(HandlerThread.java:65)

lijialiang, 2020-01-22:
I also encountered this problem : (

proxcesumit, 2023-07-24:
Got the same issue while using in fragment -
resolved by calling cameraView.close() inside onStop() of fragment instead of onPause() of fragment

proxcesumit, 2023-07-24:
Got the same issue while using in fragment -
resolved by calling cameraView.close() inside onStop() of fragment instead of onPause()