natario1/CameraView #123: Capture Documents

aliakkawi, 2017-12-13:
I am using this Repository for capturing images for documents (papers). When the paper is on a table for example, i mean the device is set horizontally above the table (device set to down), the preview image is Flipped, please advice me how to fix the issue. Thanks

natario1, 2017-12-13:
Does it happen with 1.4.0?

aliakkawi, 2017-12-13:
I am not sure, actually what i need is to capture an image for a Bank Cheque that is set on a table (the device camera should point to bottom  please check the attached images. Thanks a lot.
[image]
[image]

 ).

aliakkawi, 2017-12-14:
I am using the latest update 1.4.1

natario1, 2018-02-27:
If this is about the result, I think we can't do anything about phones in a horizontal plane position, we treat them as having screen orientation == 0, e.g. the screen natural orientation. So the result picture might be flipped with respect to what you expect. 

If this is about the preview, then it's not clear to me what the bug is. Please try to restrict the bug to a non-horizontal situation, so we can understand better what's better. The horizontal device adds confusion to this. You can open a new one. Thanks!

closed 2018-02-27 by natario1