natario1/CameraView #125: startCapturingVideo EROFS (Read-only file system)

derohimat, 2017-12-19:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [ ] *I am using the latest version*
- [ ] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. just call startCapturingVideo()

#### Expected behavior:

file can created after videoTaken

#### Actual behavior:

#### Screenshots:

#### Interesting logs:

java.io.FileNotFoundException: /blablabla.mp4: open failed: EROFS (Read-only file system)
       at libcore.io.IoBridge.open(IoBridge.java:455)
       at java.io.RandomAccessFile.<init>(RandomAccessFile.java:247)
       at java.io.RandomAccessFile.<init>(RandomAccessFile.java:128)
       at android.media.MediaRecorder.prepare(MediaRecorder.java:836)
       at com.otaliastudios.cameraview.Camera1.startVideo(Camera1.java:631)
       at com.otaliastudios.cameraview.CameraView.startCapturingVideo(CameraView.java:1195)

*You can use `CameraLogger.setLogLevel(LEVEL_VERBOSE)`
to add logs to logcat, and `CameraLogger.registerLogger()`
to handle logs yourself (e.g. export to file or a crash reporting service)
so you can collect data from your users.*

Monster-king, 2017-12-27:
You must choose correct file path which you have writing permission (don't forget to add `<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />` to your manifest)
and create file like this:
```
 File file = new File(Environment.getExternalStoragePublicDirectory("myProject"), "");
File videoFile = new File(file, "testVideo.mp4");
```

derohimat, 2017-12-28:
thats all already defined :D

natario1, 2018-01-16:
The file you are passing looks not valid, and on top of this, remember that video recording is not fully supported by emulators.

closed 2018-01-16 by natario1