natario1/CameraView #126: Camera switching front to back and visa versa are not good

LuckySambial, 2017-12-21:
All functionality working perfectly but facing issue when switching camera from front to back. Camera jumps little bit and it is not like as native camera (like flip right to left with animation).

developer--, 2017-12-25:
same for me

LuckySambial, 2017-12-26:
having this issue with samsung devices only

developer--, 2017-12-26:
Nope I tested it for Lenovo API 23

LuckySambial, 2017-12-26:
I have done with all required functionalities in project but now I am stuck with this issue only...I can not replace this lib with another because I have used lot of functionality of this library in my project....can anyone help me on this ?

natario1, 2017-12-26:
I still have no idea what this issue is about :) please post a video and the code you are using to trigger the change.

LuckySambial, 2017-12-26:
Sure I will share video with you soon
Thanks for reply!!!

LuckySambial, 2017-12-26:
please check proper issue from this video
https://www.dropbox.com/s/z7o6q8hbijtk3nq/IMG_1478.MOV?dl=0

LuckySambial, 2017-12-26:
Here is the code that I am using for toggle camera.
    private void toggleCamera() {
        if (mCapturingPicture) return;
        switch (camera.toggleFacing()) {
            case BACK:
                message("Switched to back camera!", false);
                break;

            case FRONT:
                message("Switched to front camera!", false);
                break;
        }
    }

LuckySambial, 2017-12-26:
As you can see in above video when I am trying to change camera view from front to back or visa versa then image squeezed. This is the exact issue I am facing, hope you got exact issue.

LuckySambial, 2017-12-26:
It seems device specific issue only reproduced in Samsung edge and Lenovo phone till now.

LuckySambial, 2017-12-26:
working fine with Moto and MI phone.

developer--, 2017-12-27:
I tested it also for Samsung edge and it has stretched preview, but sometimes it has correct preview. Any suggestions?

LuckySambial, 2017-12-27:
@developer I am also waiting for @natario1 response. I think he is busy in Christmas and new year party :)

natario1, 2018-02-27:
@LuckySambial so you are talking about the fact that for a small second the preview stretches before switching?

LuckySambial, 2018-02-27:
@natario1 yes exactly. can you please help me to fix that issue because my project stuck due to that issue.

kunaldawn, 2018-04-30:
Hi any update on this?

arrabidas92, 2018-09-11:
Hi, I face the same issue when switching front/back, the preview stretches before switching.
Any solution ?

closed 2019-01-04 by natario1