natario1/CameraView #129: Crash issue

protoss1010, 2017-12-28:
- [x] *I am using the latest version*
- [not] *I can reproduce the bug in the demo app, and if not...*
- [not] *I have tried to reproduce the bug in the demo app*

Fatal Exception: java.lang.RuntimeException: Fail to connect to camera service
       at android.hardware.Camera.<init>(Camera.java:519)
       at android.hardware.Camera.open(Camera.java:364)
       at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:168)
       at com.otaliastudios.cameraview.CameraController$3.run(CameraController.java:175)
       at android.os.Handler.handleCallback(Handler.java:751)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:154)
       at android.os.HandlerThread.run(HandlerThread.java:61)

muhammad-abubakar, 2018-02-12:
Did you check your manifest for appropriate permissions? Because I can only reproduce this issue If I remove <uses-permission android:name="android.permission.CAMERA" /> from manifest. Make sure CAMERA must be in all caps because I faced this issue. Secondly after 6.0 Marshmallow , you've to take care of run time camera permission as well..

natario1, 2018-02-27:
Looks like an usage issue, feel free to open a new issue if you still have this, and post exact steps to reproduce the issue and so on. I think this is better asked on StackOverflow.

closed 2018-02-27 by natario1