natario1/CameraView #131: integration issue

wajdifadool1, 2018-01-04:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [x ] *I am using the latest version*

**cant integrate the library in build.gradle 
im getting :** 

Error:Execution failed for task ':app:processDebugManifest'.
> Manifest merger failed : Attribute meta-data#android.support.VERSION@value value=(26.0.2) from [com.android.support:design:26.0.2] AndroidManifest.xml:28:13-35
  	is also present at [com.android.support:exifinterface:26.1.0] AndroidManifest.xml:25:13-35 value=(26.1.0).
  	Suggestion: add 'tools:replace="android:value"' to <meta-data> element at AndroidManifest.xml:26:9-28:38 to override.

my  buildToolsVersion is  "26.0.2"

wajdifadool1, 2018-01-04:
i Have added those in my app.gradle 

allprojects {
    repositories {
        jcenter()
        maven { url "https://jitpack.io" }
        maven { url "https://maven.google.com" }
    }

}

jslok, 2018-01-16:
in your app/build.gradle put:

    configurations.all {
       resolutionStrategy {
         eachDependency { DependencyResolveDetails details ->
           if (details.requested.group == 'com.android.support') {
              if (!requested.name.startsWith("multidex")) {
                details.useVersion '26.0.2'
              }
           }
         }
       }
    }

natario1, 2018-02-27:
I think this should be fixed now, feel free to reopen if it isn't.

closed 2018-02-27 by natario1